MagicMirrorOrg/MagicMirror #294: Custom Compliments does not work

rauchmo, 2016-05-06:
Hey :)

Why is this not working ?

PLEAS HELP xD

```
 {
			module: 'compliments',
			position: 'lower_third'
			config: {
	            compliments: {
					morning: [
						'TEST',
						'Good morning, handsome!',
						'Enjoy your day!',
						'How was your sleep?'
					]
				}
			}
		}
```

MichMich, 2016-05-06:
You forgot a `,` behind the position.

rauchmo, 2016-05-06:
Oh thanks :)
Changed it but still does not work D:

```
{
			module: 'compliments',
			position: 'lower_third',
			config: {
	            compliments: {
					morning: [
						'TEST'
					]
				}
			}
		}
```

MichMich, 2016-05-06:
try running your config.js thru a JSlinter: http://www.jslint.com

rauchmo, 2016-05-06:
THANK YOU 

i am such a idiot i forgot a (

MichMich, 2016-05-06:
Good that it's solved ... did the JSLinter help you?

closed 2016-05-06 by MichMich