MagicMirrorOrg/MagicMirror #297: Uncaught exceptions

paviro, 2016-05-06:
Is there any way we could add something, so that we know where an exception is thrown? I for example have no idea where this is coming from. A module? The core? No idea.
```
Whoops! There was an uncaught exception...
[TypeError: Cannot read property 'name' of undefined]
MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
```

KirAsh4, 2016-05-06:
The thingamabob couldn't talk to the thingamajig, and therefor the whozits threw a fit which the whatzits reported to you. Sorry. :)

MichMich, 2016-05-10:
Since this is not really a bug, but has more to so with the further development of the core, I think we should discuss this on the forum (so I can keep the issue list) clean.

closed 2016-05-10 by MichMich