MagicMirrorOrg/MagicMirror #301: Add developer mode

planetmadness, 2016-05-08:
It seems that, by default, the rightclick menu is hidden, which prevents the use of inspect element. The cursor is hidden by default as well.
I completely understand that this makes sense, however, when trying to apply custom styling and debugging it is quite a setback. Perhaps there could be a developer mode added as a flag for example, which enables the rightclick menu and does not hide the cursor.

Another addition could be another flag such as a no-reload flag which loads the data once and does not reload. This prevents that you are manipulating an element with the developer tools and all of a sudden it's all gone :(

MichMich, 2016-05-08:
For the first part of your question, you could just use the custom.css. Also, you might want to dicuss issues like this on http://forum.magicmirror.builders 👍

planetmadness, 2016-05-08:
Good to know! Wasn't aware that the forum is already live, last time I checked it was still in development.

KirAsh4, 2016-05-08:
There's another way. But I'll wait for you to open a topic on the forum instead.

MichMich, 2016-05-10:
Since this is taken to the forum, I close this issue. If you want me to reopen. Please let me know.

closed 2016-05-10 by MichMich

jtyle6, 2016-05-10:
Control+Shift+I on the pi that is running MM and that pops up Developer tools. 
[image]