MagicMirrorOrg/MagicMirror #302: White screen on page reload

planetmadness, 2016-05-08:
While customizing and testing stuff, I need to reload sometimes. Everytime I do this, or if I restart the mirror the entire screen goes white. I don't know if this is possible to change, as it is the browser. But if it would be possible to get the browser to display a black screen that would be better!

MichMich, 2016-05-08:
That's not possible to change since its part of electron. My suggestion is to use 'node serveronly' for testing and just use your browser. This was the reason why I added it ...

paviro, 2016-05-08:
Maybe look at [this](https://github.com/electron/electron/issues/861)?

MichMich, 2016-05-10:
I've got currently no plans to work in this, since restarting the MM is not something you usually do regularly. For developers, I suggest going for the `node serveronly` route.

Of course, a pull request to modify the white screen is welcome.

closed 2016-05-10 by MichMich