MagicMirrorOrg/MagicMirror #314: Can't parse feed item in log?

KirAsh4, 2016-05-11:
**Platform**: Linux

**Node Version**: node 6.0.0

**MagicMirror Version**: v2-beta

**Description:**: Fresh clone of the v2-beta repository, running `node serveronly` and this is what it spits out on screen after loading the mirror (only putting a few blocks as they're all the same error):

    Can't parse feed item:
    { text: '\n\n\n\n\n\n\n\n',
      title: 'Why we need a cyber wall',
      guid: 'http://www.cnn.com/2016/05/02/opinions/build-cyber-wall-vishwanath/index.html',
      link: 'http://rss.cnn.com/r/66bd6a36da/0L0Scnn0Bcom0C20A160C0A50C0A20Copinions0Cbuild0Ecyber0Ewall0Evishwanath0Cindex0Bhtml0Deref0Frss0Itech',
      description: '',
      pubdate: 'Tue, 03 May 2016 08:33:22 EDT',
      'media:thumbnail': 
       { height: '51',
         width: '90',
         url: 'http://i2.cdn.turner.com/cnn/dam/assets/140806080947-cyber-attack-file-top-tease.jpg' },
      'media:content': 
       { height: '51',
         width: '90',
         medium: 'image',
         type: 'image/jpeg',
         url: 'http://i2.cdn.turner.com/cnn/dam/assets/140806080947-cyber-attack-file-top-tease.jpg' } }
    Can't parse feed item:
    { text: '\n\n\n\n\n\n\n\n',
      title: 'How real is hypersonic aircraft revolution?',
      guid: 'http://www.cnn.com/2016/03/07/aviation/hypersonic-future-of-aviation/index.html',
      link: 'http://rss.cnn.com/r/66bd6a36da/0L0Scnn0Bcom0C20A160C0A30C0A70Caviation0Chypersonic0Efuture0Eof0Eaviation0Cindex0Bhtml0Deref0Frss0Itech',
      description: '',
      pubdate: 'Wed, 09 Mar 2016 09:58:49 EST',
      'media:thumbnail': 
       { height: '51',
         width: '90',
         url: 'http://i2.cdn.turner.com/cnn/dam/assets/160303161943-tznasamontage-top-tease.jpg' },
      'media:content': 
       { height: '51',
         width: '90',
         medium: 'image',
         type: 'image/jpeg',
         url: 'http://i2.cdn.turner.com/cnn/dam/assets/160303161943-tznasamontage-top-tease.jpg' } }

MichMich, 2016-05-11:
Node 6 is not supported. (Should've somewhere in the docs.) it appears a lot of modules still have issues in node 6. PR to fix it is welcome.

KirAsh4, 2016-05-11:
Well unfortunately, for many who already have node 6 installed, that means having to figure out how to downgrade it back. However, I just thought of something ... when I reinstalled (by running the installer curl call), that should've grabbed the earlier version of node. Doesn't it?

MichMich, 2016-05-12:
Give it a try. :)

MichMich, 2016-05-12:
I just tried Node 6.1.0 and it seems to work fine. What's the news feed URL that gave the errors?

jtyle6, 2016-05-12:
I think it's http://edition.cnn.com/services/rss/

MichMich, 2016-05-12:
This appeared to be an issue with the CNN feed. Some items were missing a description. This is fixed now.

closed 2016-05-12 by MichMich