MagicMirrorOrg/MagicMirror #319: Null displayed using Microsoft EDGE browser

benjeefox, 2016-05-12:
**Platform** 
Raspberry Pi 2 as serveronly
Browser client Microsoft EDGE from Windows 10

**Node Version**
v5.9.1

**MagicMirror Version** 
v2-beta

**Description:** 
While browsing MM2 on Microsoft EDGE, some NULL are displayed in each module

For screenshot:
[https://forum.magicmirror.builders/topic/78/null-displayed-using-microsoft-edge-browser](url)

KirAsh4, 2016-05-18:
I don't want you to think this is being ignored. I see it too on my end ... I just don't use Edge but I don't expect everyone to do that. :)

I'm working on some other code (and different projects), but i will try to find some time and dig into this. One of the immediate things that comes to mind is that the Edge browser is still very new and doesn't have the same compatibility as the more mature browsers (I use Chrome, Firefox, and Internet Explorer, in that order of preference), so it's likely this is something with the browser itself that we can't do anything about. However, nothing is for sure till we investigate.

Just as an FYI, on a Windows 10 machine, you do still have Internet Explorer, they just hide it. But you can still use that instead of the newer Edge browser.

cschwarz007, 2016-06-18:
Note I have the same issue (diesplaying "null") in IE11 on my Win10 machine. But I can confirm the issue does not appear with Google Chrome (on Win 10). See attached screenshot for examples...

[image]

KirAsh4, 2016-06-19:
Yes, this is specific to Edge. Being that it's a new browser, there is still a lot missing in it. Use Internet Explorer instead (yes it is installed, you aren't forced to use Edge.)

KirAsh4, 2016-07-01:
This has now manifested itself in the most recent version of Microsoft Explorer as well. I tracked it down and submitted a PR for it, #366 ...

KirAsh4, 2016-07-10:
@benjeefox and @cschwarz007, have you had a chance to pull the dev branch and test this, see if the bug still exists?

cschwarz007, 2016-07-13:
Hi, I can confirm PR #366 fixes the issue... Thanks a lot!

closed 2016-07-22 by MichMich

ljw98, 2019-02-16:
hello,i want to know how can i run this in windows10 with Google Chrome

justjim1220, 2019-02-16:
Node serveronly

sindhuja1999, 2022-01-13:
how to get access token and bearer authentication in smart mirror?