MagicMirrorOrg/MagicMirror #320: bash: line 1: Not: command not found

Labzz, 2016-05-12:
Hi dear,

First at all, im using the Raspberry pi 2, with Rasbian Jessy (complete image)  
Does someone else got the same error using the following command? Or does anyone knows how to fix it? yesterday it still worked, but now i stuck. 
" curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh | bash "
I already reinstalled rasbian multiple times, and troubleshooted on internet.

Hope to hear you soon, thanks anyways. 
[image]

MichMich, 2016-05-12:
What do you see when you type 

    curl https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh

Labzz, 2016-05-12:
The whole code. 
[image]

Labzz, 2016-05-12:
[image]
(Without;  | bash)

MichMich, 2016-05-12:
In your screenshots (where it fails) You are using the wrong url. /master/ is missing in the url.

closed 2016-05-13 by MichMich