MagicMirrorOrg/MagicMirror #327: load json file

mastrom90, 2016-05-13:
Hi guys,
i have a really little problem and i think it is easy to be solved. I have created a  new module where i need to load a local json file which is created by an external program.
I cant figure out how i can read this file and where to locate that one. I have created a variable with the filetext so i know i can handle it if i read it. But i cant get it to read the file so i can remove the variable because the data changes every week.

i hope someone can help me. Sorry if im wrong here but i dont know how i can help myself :(

paviro, 2016-05-13:
For development help check the forum: http://forum.magicmirror.builders

mastrom90, 2016-05-13:
thank you for that

closed 2016-05-13 by MichMich