MagicMirrorOrg/MagicMirror #328: Install MM² on a ARMv8 x64

djsunrise19, 2016-05-14:
**Platform** [ Odroid C2 with Ubuntu Mate 16.04 LTS]:

**Node Version** [ 6.1 (newest version) ]:

**MagicMirror Version** [ V2 ]:

**Description:** Last week I bought an odroid c2 and I wanted to check if the MM² will work on it. As I expected, the installer script won't work. So I installed Node 6.1 for ARMv8 manually, `git pull` the V2 branch and `npm install`. There is an electron error, he cannt find the file to download. Is there a way for doing the last steps manually? Log is under "configuration" (scroll at the bottom in this file).

**Steps to Reproduce:** `npm install` on a ARMv8.

**Expected Results:** -/-.

**Actual Results:** -/-

**Configuration:** 
[npm-debug.txt](https://github.com/MichMich/MagicMirror/files/264450/npm-debug.txt)

**Additional Notes:** -/-

nhubbard, 2016-05-17:
It looks like Electron isn't prebuilt on this platform. @MichMich should probably add the enhancement tag.

MichMich, 2016-05-17:
Indeed, Electron does not support this platform. Please open an issue in the electron repository.

djsunrise19, 2016-05-17:
Thank you. I reported an issue: https://github.com/electron/electron/issues/5558

MichMich, 2016-05-18:
I'll close this issue since it's an electron issue. If you want me to reopen it, feel free to let me know.

closed 2016-05-18 by MichMich

sdetweil, 2018-05-09:
i have an odroid as well, it runs another electron based smart mirror without problem. 
i get a black screen, trying to npm install in vendor fails with 
```
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@^1.0.0 (node_modules/chokidar/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
```