MagicMirrorOrg/MagicMirror #331: Newsfeed appears to crash the electron app - no error message(s)

alphapixie, 2016-05-16:
Appears that the newsfeed module is making the electron app shutdown - app silently shuts down.  I tried to change the logging levels but didnt make a difference in terms of useful errors, etc.  I confirmed this issue is specific to the newsfeed by configuring each module one at a time.  Likewise, I tried several different feed and module configuration options.  

```
    {
        module: 'newsfeed',
        position: 'bottom_bar',
        config: {
            feeds: [
                {
                    title: "New York Times",
                    url: "http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml"
                },
                {
                    title: "BBC",
                    url: "http://feeds.bbci.co.uk/news/video_and_audio/news_front_page/rss.xml?edition=us"
                }
            ]
        }
    }
```

nodejs install source:
http://thisdavej.com/beginners-guide-to-installing-node-js-on-a-raspberry-pi/
(specifically this version of the install:  https://github.com/nodesource/distributions/blob/master/deb/setup_6.x)
nodejs version: v0.6.19
npm version: 1.1.4

MichMich, 2016-05-17:
Could you try updating Node by running the MagicMirror installer script?

alphapixie, 2016-05-18:
What version of Node should I be running?  What is the best place to get this version for a v2 Pi?

MichMich, 2016-05-18:
At least 5.1.0 (this is what I've tested.)
But if you run the MM installer, it will be upgraded automatically. Just run:
```
curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh | bash
```

For further support, please check http://forum.magicmirror.builders

closed 2016-05-19 by MichMich