MagicMirrorOrg/MagicMirror #332: Node ServerOnly Autostart

cyberdie, 2016-05-17:
I'm trying to setup MagicMirror on a Raspberry B+
Apart from high cpu usage of npm i'm in trouble starting node with serveronly parameter on startup.
The effect is that Midori cannot open http://localhost:8080 because node is not started.
If i connect via SSH, launch "node serveronly" and reload Midori (via vnc) all works fine.

How could i start node on startup with "serveronly" parameter?

MichMich, 2016-05-17:
Just run `node serveronly` from the MagicMirror folder.

Even though this should work, It's not officially supported. Further support questions can be asked on the forum: http://forum.magicmirror.builders

Cheers, Michael.

closed 2016-05-17 by MichMich

cyberdie, 2016-05-17:
Thanks Michael but what if i restart the Pi?
How can i automate "node serveronly" command?
I've tried on X Autostart without success..

MichMich, 2016-05-17:
See https://github.com/MichMich/MagicMirror/wiki/Auto-Starting-MagicMirror