MagicMirrorOrg/MagicMirror #338: Multiple RSS Newsfeeds in config.js

GlenNicholls, 2016-05-22:
Hey,

Currently, I am trying to add multiple newsfeeds for the newsfeed module. I am using identical syntax to that of the example you have under the tutorial. However, when I run the mirror on my port, it only shows whichever newsfeed comes first. I have tested them both on their own to make sure they are the correct format of xml, and they work fine on their own. I have tried putting them in different places on the screen to see if they were just getting cut off. After reading through the readme and your notes in the code, I do not see where I am going wrong. Everything for the newsfeeds is default with the exception of adding one more newsfeed. Do you have something set up where they are going to alternate or change which newsfeed it displays after a certain time or if they´re supposed to be displayed simultaneously? What can I change to allow me to display both simultaneously or alternating?

GlenNicholls, 2016-05-22:
EDIT:
After looking through it again, should I change the reloadInterval to a smaller amount of time to allow it to actually alternate between the two?

MichMich, 2016-05-22:
Please ask any support question on http://forum.magicmirror.builders 👍🏻

closed 2016-05-22 by MichMich

pieroway, 2018-10-07:
I'll give my config file entries. which worked first time to display both at the bottom of the screen.
This is for anyone that comes here to read this in the future.
```
                        ... 
                },
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "Global News",
						url: "https://globalnews.ca/ottawa/feed/"
					}
				],
				showSourceTitle: true,
				showPublishDate: true
			}
		},
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "New York Times",
						url: "http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml"
					}
				],
				showSourceTitle: true,
				showPublishDate: true
			}
		},
                { 
```
                     ...