MagicMirrorOrg/MagicMirror #339: OpenWeather unambiguous locations

mochman, 2016-05-25:
In the default currentweather and weatherforecast modules you use a location name for your searching.  I have an update that will look to see if a user puts in a [locationID](http://bulk.openweathermap.org/sample/).  The new getParams() function will handle the request.  If the user doesn't put in a locationID, it will default to the typed out location.

Is this something you'd like me to make a pull request for?

MichMich, 2016-05-25:
Yes, please!

mochman, 2016-05-25:
Okay, now since this is my first time using git hub to add code to someone else's project, would you mind letting me know how to?  Or I can just send you the new .js files directly.

MichMich, 2016-05-26:
Thanks! Merged to `develop`.

closed 2016-05-26 by MichMich