MagicMirrorOrg/MagicMirror #354: magicmirror2 'does' work on the old model-B pi

vinceskahan, 2016-06-05:
Just wanted to comment that you 'can' get MagicMirror2 working pretty simply on the original model-B  if you use the version of node from http://node-arm.herokuapp.com

I used the wget method to download it and dpkg to install it.  The resulting 'npm install' per your instructions took a 'long' time, but completed fine.

I'm running Raspbian 2016-05-27 just as a point of reference.

MichMich, 2016-06-06:
Does Electron run as well?

vinceskahan, 2016-06-06:
how could I tell ?

I'm running serveronly but can hook it up to a monitor if you need me to
check it that way...

jtyle6, 2016-06-06:
Go to your pi ip address with Web browser with. 
```
:8080
```
At the end so it looks like xx.x.x.xx:8080

paviro, 2016-06-06:
Server only was never a problem wasn't it?

KirAsh4, 2016-06-06:
I suspect you'll have issues with Electron on an rpi-modelB, but till you hook up a monitor and try (re)starting it, there's no telling.

@jtyle6, going to the address like that does not actually do anything as you're using a completely different browser to do that. Electron runs on the rpi itself and is visible on a monitor connected to it.

jtyle6, 2016-06-07:
Does not work on a B and B+ this is the same image as Pi 3.

``` 
node serveronly                                                      132 ↵
[1]    1381 illegal hardware instruction  node server only
```

@KirAsh4 fixed the ip.

AntiElephantMine, 2016-06-20:
Do you have display issues running this on your B+? I have exactly the issue mentioned here:

https://forum.magicmirror.builders/topic/80/chromium-rpi-1-display-strange

where the newsfeed and compliments won't go to the bottom. Unfortunately I'm not very skilled at CSS so I can't see the problem.

Fabrice5800, 2016-07-08:
Hello, i'm french i try to install the mirror 2 but i have more problems !!!
HELP PLEASE   
npm ERR! Linux 4.4.11-v7+
npm ERR! argv "/usr/local/bin/node" "/usr/bin/npm" "start"
npm ERR! node v4.2.1
npm ERR! npm  v3.10.3
npm ERR! path /home/fablab/package.json
npm ERR! code ENOENT
npm ERR! errno -2
npm ERR! syscall open

npm ERR! enoent ENOENT: no such file or directory, open '/home/fablab/package.json'
npm ERR! enoent ENOENT: no such file or directory, open '/home/fablab/package.json'
npm ERR! enoent This is most likely not a problem with npm itself
npm ERR! enoent and is related to npm not being able to find a file.
npm ERR! enoent 

npm ERR! Please include the following file with any support request:
npm ERR!     /home/fablab/npm-debug.log

closed 2016-07-22 by MichMich