MagicMirrorOrg/MagicMirror #357: RSS errors ... still

KirAsh4, 2016-06-06:
_(Going to be thorough here. :) )_

**Platform:**
Raspberry Pi 3, running Rasbian-Jessie.

**Node Version:**
Node 6.0.0

**MagicMirror Version:**
v2.0.2

**Description:**
NY Times RSS feed (and others as well) will occasionally throw an error about being unable to parse a specific item.

**Steps to Reproduce:**
Refresh the MM task, or just wait for the feed to cycle through to that news item again.

**Expected Results:**
No errors!

**Actual Results:**
Log file shows this:
```
Can't` parse feed item:
{ title: '',
  link: 'http://www.nytimes.com/interactive/2016/06/05/world/americas/panama-papers-us-taxes.html?partner=rss&emc=rss',
  guid: 
   { ispermalink: 'true',
     text: 'http://www.nytimes.com/interactive/2016/06/05/world/americas/panama-papers-us-taxes.html' },
  'atom:link': 
   { rel: 'standout',
     href: 'http://www.nytimes.com/interactive/2016/06/05/world/americas/panama-papers-us-taxes.html?partner=rss&emc=rss' },
  description: '',
  'dc:creator': '',
  pubdate: 'Mon, 06 Jun 2016 16:30:06 GMT' }
Title: 
Description: 
Pubdate: Mon, 06 Jun 2016 16:30:06 GMT
```

**Configuration:**
```
                {
                        module: 'newsfeed',
                        position: 'bottom_bar',
                        config: {
                                feeds: [
                                        {
                                                title: "New York Times",
                                                url: "http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml"
                                        },
                                        {
                                                title: "CNN: Technology",
                                                url: "http://rss.cnn.com/rss/edition_technology.rss"
                                        }
                                ],
                                showSourceTitle: true,
                                showPublishDate: true
                        }
                },
```

**Additional Notes:**
Because on a normal-running MM setup, these log notifications go in `~pi/pm2/logs/MagicMirror-out-0.log`, that file will grow rather fast on a mirror that's been running for days. I just deleted mine (through `pm2 flush`) which was over 2Gb in size. A regular user might not know about this or how to manage their pm2 logs.

MichMich, 2016-06-06:
It seems it doesn't get a title from the NYTimes for that item. Therefor it can't parse it. This only means it won't show up on your mirror. Which is logical, since there isn't a title to show.

KirAsh4, 2016-06-06:
Those college interns, sheesh. One job, that's all they have. :)
Rather than filling up the log file with all that info, maybe it's time to think about adding a debugging flag. By default only put a single line in the log file "This-and-that failed, call 1-800-Michael." And if you need more info, flip a debug flag in the config and reload so it would spit out more info. This just fills up the log file ... And because I have (at least) two feeds, my log file filled up faster than I could say 'Oh shit!' I was surprised to see my rpi's storage at 99% yesterday ... that one log file was huge.

MichMich, 2016-06-06:
That's no good indeed ... Unfortunately, I did not use a log alias in the node_helper like I did in the fronted using Log.log. This would make it very simple to implement ...

So: looking forward to your PR! ;)

(I'll put it on my ToDo list.)

kamranrana, 2016-06-27:
If I wanted to add an ESPN RSS feed to my MagicMirror, how would I be able to do so using the config file and the node.js file?

MichMich, 2016-06-27:
@kamtanrana Here's everything You need to know: https://github.com/MichMich/MagicMirror/blob/master/modules/default/newsfeed/README.md

For future support questions, please check out forum: http://forum.magicmirror.builders

KirAsh4, 2016-06-27:
Hey @MichMich, do those 'extension-less' feeds work? None of ESPS's feed end in either .rss or .xml.

MichMich, 2016-06-27:
As long as it's a xml or json feed. The extension doesn't matter. It the contents what counts.

KirAsh4, 2016-06-27:
Good deal.

kamranrana, 2016-06-27:
@MichMich That doesn't really work, I've followed the README step by step and even tried the basic BBC rss feed and all it displays on the mirror is Loading...           Maybe it's not being parsed correctly?

MichMich, 2016-06-27:
Please open a topic on the forum. Thanks.

KirAsh4, 2016-06-27:
Hey @kamranrana, would you mind taking it to the Forum where others can also help?

kamranrana, 2016-06-27:
@KirAsh4 The website says "Confirm your email" and when I click there it won't do anything, and it says that it has lost connection to the website

KirAsh4, 2016-06-27:
That's a call to @paviro ... out of curiosity, try a different browser and see if you get the same result.

MichMich, 2017-01-15:
I close this issue for now. Feel free to reopen.

closed 2017-01-15 by MichMich