MagicMirrorOrg/MagicMirror #363: Alternate webserver

jsaddiction, 2016-06-14:
Have you ever thought of creating a branch of this web app designed to reside on an NAS or similar webserver?  My thought process on this concept is to enable the use of any web capable device to reproduce the webpage and display it.

I came across the Magic Mirror idea while browsing for new projects and several days later I decided that once I am back from deployment I want to recreate your project and possibly write some more modules.  I thoroughly enjoy the "module" concept and allowing 3rd parties to create extra content.  This allows many different configurations and customizations.  By far I am interested in your project.
Well Done!!

The reason I post a question like this is purely from the aspect of enabling more users and allowing more devices to act as the display.  For instance you could buy an old tablet online and mount it behind some glass and you have a Magic Mirror.

I personally own a Synology NAS and have already downloaded a zip version of your application and began playing around with some of the code.  I will admit I am a NOOB at web design but this application has inspired me to learn.  

Let me know what you think.  I would be willing to take on this project but in doing so I might need some help at least until I learn the ins and outs of your design.

paviro, 2016-06-14:
You could always run the serveronly version on a NAS if there is node js for it :)

jsaddiction, 2016-06-14:
I was looking through the "serveronly" js and apparently I don't see how to implement that function into a server side application.  Please excuse my ignorance on the functionality of this application but to utilize my NAS as a host for this application I think I still need some sort of html request to initiate all of the js.  Could you further explain the process?

MichMich, 2016-06-14:
Hi @jsaddiction, the "serveronly" variant is a web server itself, without any UI (electron). As far as I understand your initial question, this "serveronly" tool is all you need on your NAS.

Make sure you have a recent version of Node.js installed in your NAS (at least 5.10.0), clone the MagicMirror repo, run `npm install` and then you are ready to run the server on your NAS: `node serveronly`

After you started the server you can access the UI using any computer on your network via: `http://ipaddressofyournas:8080`

jsaddiction, 2016-06-14:
@MichMich To be more specific, my nas already runs a web server (apache).  I was more inquiring the ability to utilize the front end of this application with the already installed web server instead of the node.js/serveronly solution.  Looking through the code and console messages I think and please correct me if I am wrong, some of the js you have requires the use of node.js.  I can get the clock working and complements working without the use of node.js since it doesn't use any socket connections and thus never uses any node.js calls..  How difficult would it be to reproduce your functionality without a node.js dependency, and still retain module compatibility?

MichMich, 2016-06-14:
The node web server also facilitates the socket server for communication with the backend. This communication is necessary for most of the (more advanced) modules like news & calendar (an a lot of 3rd party modules). If you would want to run the app using your own web server, this would mean you have to split the sockets server from the web server. Besides that, the MagicMirror API allows the developers to leverage the built in web server. Which means you have to rebuilt that part of the API as well ...

If the question is: is this possible? Then the answer is: yes, if you have a lot of spare time ... ;) 

But honestly, it really wouldn't make any sense. Why not just run both servers side by side? You can he change the port of the Magic Mirror server to whatever port you like in the config.

jsaddiction, 2016-06-14:
You make a bunch of very valid points and helps clear up the mud so to speak.  I suppose I was thinking of keeping inline with your modular design and allow both use of embedded webserver or a roll your own implementation.  

Spare time isn't always something I have a large reserve of.  However, I do have my hobbies and create as much time as I can.  Would you be apposed to me "stealing" some of your design concepts and more or less piggy back off your project?  Would you also be able to assist (point me in the right direction) occasionally as I build and learn this?  Like I mentioned, I would like to retain full compatibility of the 3rd party modules.  (This is more of a learning experience for me)

MichMich, 2016-06-16:
Since MagicMirror has an [MIT License](https://github.com/MichMich/MagicMirror/blob/master/LICENSE.md) you're free do "steal" any of the code you want. Unfortunately I'm unable to assist you due to my limited amount of spare time (i'd rather use my time to improve the MagicMirror platform.)

That being said, If you want to learn stuff like this, why not invest your time in creating native MagicMirror modules? This way, anyone can benefit of your development. And you can get the support you want on our forum.

Check out the module builders section on our forum: https://forum.magicmirror.builders/category/5/modules

Cheers!

jsaddiction, 2016-06-16:
Thank you!  I do not currently have an internet connection for my development platform but when I get the time and access I will look into a module or two and put a couple of repos up for everyone.  I appreciate the time you have taken to consider my questions!

closed 2016-06-16 by jsaddiction