MagicMirrorOrg/MagicMirror #364: Error: Cannot find module 'express'

baba-123, 2016-06-21:
Installed on PI-1. getting the following error when running "node serveronly"

pi@raspberrypi:~/MagicMirror $ node serveronly

module.js:340
    throw err;
          ^
Error: Cannot find module 'express'
    at Function.Module._resolveFilename (module.js:338:15)
    at Function.Module._load (module.js:280:25)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/home/pi/MagicMirror/js/server.js:8:15)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)

MichMich, 2016-06-21:
Sorry, Pi 1 is not supported.

closed 2016-06-21 by MichMich

baba-123, 2016-06-21:
Is there a basic version I can install...I have come a long way with
setting up all the hardware and to stop because I have the wrong version of
pi is sad....

Any work around??

MichMich, 2016-06-21:
You might be able to find some workarounds on http://forum.magicmirror.builders - But unfortunately I can't give you any support.

baba-123, 2016-06-21:
just to confirm, I am using Raspberry Pi Model B 512MB RAM. can you confirm
is this not supported?

MichMich, 2016-06-21:
Exactly. Officially only the Pi 2 & 3 are supported.

baba-123, 2016-06-21:
Thank you for your assistance...

chenz0, 2016-06-21:
Officially, Pi 1 supported. I was able to get it running on my Pi 1 though on serveronly. The challenge was getting the layout right in the browser. I've since had to reformat my RPi so I can't tell you much more but it is possible. 

There's another thread discussing how to run on Pi1: 
https://github.com/MichMich/MagicMirror/issues/354

I haven't gotten this method to work yet though

chenz0, 2016-06-21:
*is not supported. Sorry

baba-123, 2016-10-28:
Thank you for you help. I was able to build my own magic mirror...Additionally I add a sensor to control the monitor, which will save power and also not overheat the system...

much appreciated..attached are some pictures

[image]
[image]
[image]
[image]
[image]