MagicMirrorOrg/MagicMirror #367: newsfeed module doesn't read rss-feed

CyberClwn, 2016-07-03:
**Platform**
Raspberry Pi 2 (wheezy)

**Node Version**
v4.1.0

**MagicMirror Version**
V2

**Description:**
If i put this in the config.js to use the newsfeed-module, the screen keeps on saying "Bezig met Laden ..." and doesn't show any information from this feed:
{
                        module: 'newsfeed',
                        position: 'bottom_bar',
                        config: {
                                feeds: [
                                        {
                                                title: 'Blizzard',
                                                url: 'http://eu.battle.net/wow/en/feed/news'
                                        }
                                ]
                        }
                }
If i specify another feed it does work, so my configuration-lay out is ok. (eg. "http://www.ad.nl/home/rss.xml").
I'm not sure if it is the feed or the module, since i can't find another feed that doesn't specify a filename.
The rest of the modules load and do show information (weather and calendars).

This is the error that is shown:
Whoops! There was an uncaught exception...
TypeError: description.replace is not a function
    at .<anonymous> (/home/pi/MagicMirror/modules/default/newsfeed/fetcher.js:53:31)
    at emitOne (events.js:96:13)
    at emit (events.js:188:7)
    at SAXStream.wrapper.__wrappers.(anonymous function) (/home/pi/MagicMirror/node_modules/feedme/node_modules/eventyoshi/lib/index.js:135:12)
    at emitOne (events.js:96:13)
    at SAXStream.emit (events.js:188:7)
    at SAXStream.onclosetag (/home/pi/MagicMirror/node_modules/feedme/lib/xmlfeedparser.js:140:14)
    at emitOne (events.js:96:13)
    at SAXStream.emit (events.js:188:7)
    at Object.me._parser.(anonymous function) [as onclosetag] (/home/pi/MagicMirror/node_modules/feedme/node_modules/sax/lib/sax.js:258:17)

MichMich, 2016-07-04:
This is probably because You use an old Node version. Run the installer to upgrade node. (You might also need to upgrade to Jessie)

CyberClwn, 2016-07-04:
I was afraid i would get an easy answer like that.
The Pi is running more (background-) software and i can't easily upgrade it (to Jessie).
But since the other feed is working i doubt it is the Node version.
The installer didn't even work on Wheezy, so i had to install some software myself, to get this far. 
Was hoping for some more support. But i'll look for another solution then.

MichMich, 2016-07-04:
Wheezy isn't officially supported, so I can't help you in a github issue. You might have better luck on the support forum: http://forum.magicmirror.builders

ettisan, 2016-07-14:
I'm having a similar problem, newsfeed and calendar show "Loading"
I've used the Chrome console to debug the communication (is there any way to get verbose output in "serveronly" mode?) and found something like this in the server response (inside the binary socket.io stuff):

/newsfeed,"Invalid namespace"
/calendar,"Invalid namespace"

I don't know what namespaces cause the issue (XML namespaces, are there JS namespaces?). The sample feed (NYT) doesn't work either.

ettisan, 2016-07-14:
ah, just figured out those are probably socket.io namespaces

MichMich, 2016-07-14:
What version of Node are You running?

ettisan, 2016-07-14:
right.... sorry. thought I had upgraded node - apparently I didn't. works fine now with v6.2.2

MichMich, 2016-07-14:
Good to hear! :)

closed 2016-07-22 by MichMich