MagicMirrorOrg/MagicMirror #368: Getting 502 error with npm

Anool, 2016-07-06:
running the automatic script fails with 
"Error: fetch failed with status code 502"

Debug file attached. npm is having issues today, so I'm wondering if there is a way to temporarily fix this.

[npm-debug.log.txt](https://github.com/MichMich/MagicMirror/files/349747/npm-debug.log.txt)

Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: https://forum.magicmirror.builders/category/15/bug-hunt - Problems installing or configuring your MagicMirror? Check out: https://forum.magicmirror.builders/category/10/troubleshooting

**Platform** [ Raspberry Pi 2/3, Windows, Mac OS X, Linux, Etc ... ]:

**Node Version** [ 0.12.13 or later ]:

**MagicMirror Version** [ V1 / V2-Beta ]:

**Description:** Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce:** List the step by step process to reproduce the issue.

**Expected Results:** Describe what you expected to see.

**Actual Results:** Describe what you actually saw.

**Configuration:** What does the used config.js file look like? (Don't forget to remove any sensitive information.)

**Additional Notes:** Provide any other relevant notes not previously mentioned (optional)

MichMich, 2016-07-06:
`error fetch failed with status code 502`

Probably an issue with NPM indeed.

nhubbard, 2016-07-06:
Hi @MichMich, I'm back! Yes, it is an issue with NPM. I was having issues with NPM earlier today, and the best way to fix it is to wait. It's only happening in select regions, and they deployed a temporary patch as of 14:54 UTC.

Anool, 2016-07-07:
Fixed and works now.

closed 2016-07-07 by Anool