MagicMirrorOrg/MagicMirror #374: No Pi Zero support

jfi1310, 2016-07-15:
Please add support for the Pi Zero.

MichMich, 2016-07-15:
I think it should run on the Zero. I did not try it though. What errors are you experiencing?

jfi1310, 2016-07-15:
While trying to install it, there was this message:

[image]

MichMich, 2016-07-15:
What do you get when you run the following command?
```
uname -a
```

jfi1310, 2016-07-15:
That's what I get:
`Linux raspberrypi 4.4.11+ #888 Mon May 23 20:02:58 BST 2016 armv6l GNU/Linux`

MichMich, 2016-07-15:
Hmmm ... it seems the Zero is running on `ARMv6`. Just like the the Pi 1. Unfortunately, Electron (the browser frame on which MagicMirror relies) does not support `ARMv6`. You can install MagicMirror² and just use the `serveronly` method, but you would need to setup a browser in kiosk mode yourself. Which is tricky on the `ARMv6` platform.

kapsolas, 2016-07-15:
I successfully installed it on a Zero. 

I had to get the Electron build for the arm processor i'm using. Its available now.

MichMich, 2016-07-20:
@kapsolas How did you install electron, simply using NPM? Or did you need to manually compile?

kapsolas, 2016-07-21:
To be honest, I don't remember. I tried a few things before i got it. I did end up though on the electron site where I downloaded from there an arm build. Think this was the one that worked.

MichMich, 2016-07-21:
Ok, then I need to het me a Pi zero to test it. Till then I can't add support. Thanks for the info!

kapsolas, 2016-07-21:
No worries!

nhubbard, 2016-09-07:
We probably should close this issue, as it was solved by commit af279df5c9845b3063d906b441ddcddef9a4493c.

closed 2016-09-07 by MichMich

alessandroiacovella, 2018-07-19:
can you make a one-line command that makes the pi zero w work like the one that is only compatible with pi 2 and 3 or can you make it easier for us to understand server only please 

Thanks

commit a2497d75 (2016-09-07): Merge pull request #430 from nhubbard/develop

Close #374
commit af279df5 (2016-09-07): Closes #374.

ARMv6 is not supported by recent versions of Chrome, and by extent, this means recent versions of Electron as well.