MagicMirrorOrg/MagicMirror #376: General Question on updates and modules

kapsolas, 2016-07-19:
Hi,

I have the greek translation ready to submit, but I am not sure. What is the process for providing this to you to incorporate into your master branch? (Not a github expert by any means)

Additionally. I have written a Flickr module, how can that be added to the list of modules available?
[MMM-Flickr] (https://github.com/kapsolas/MMM-Flickr)

Thanks!

KirAsh4, 2016-07-20:
Please sign up to the forum at https://forum.magicmirror.builders/ and post your questions there for more help. In the mean time, here's a run-down of what you need to do:

First put your code up on GitHub, then submit a PR request for the translation file against the develop branch. Do it in steps (and ask help along the way):
- Fork the main MagicMirror² project into your own account
- Push your changed/added files to the 'develop' branch
- Submit a PR request against the 'develop' branch

If all that sounded foreign, well it kinda is. Github has a learning curve. But on this page that you are reading, at the top on the right, you should see a button that says 'Fork' with a number behind it. Click on it. That will fork the project into your own account.

Once forked into your account, change the branch to 'develop'. Look at the top of this page, right below your username/MagicMirror, and you should see a menu, 'Code, Issues, Pull requests, etc., etc'. Make sure the 'Code' one is selected. A little below that, there's a button that read 'Branch: master'. Click on it and select the 'develop' one.

Now you can use Github's editor and manually post and commit your changed files into the branch, or you can use a facility such as 'git' on your own computer to clone the whole thing to your computer, add/change the files there, then push it all back up to Github. Either way, this is a more involved process and I recommend you research it some more using Github's guides at https://guides.github.com/activities/hello-world/ (main page is at http://guides.github.com/).

Once your changed code is on Github, you can create a pull-request (PR) against the same 'develop' branch and submit it.

For the Flickr module you create, you need to create your own Github repository and post it on the forum so others can find it/talk about it/etc., etc.

Hope this helps.

kapsolas, 2016-07-21:
I did not have any success logging into the forum. I kept getting an error that said invalid session no matter what I did, thus I decided to post here. 

The link to the module is already on my github. (MMM-Flickr)[https://github.com/kapsolas/MMM-Flickr]

MichMich, 2016-07-21:
Hi @kapsolas. That login issue should be solved now. @paviro made some changes to the forum earlier this week. You might need to to remote your cookies.

MichMich, 2016-07-22:
If you have any additional questions, feel free to ask. I'll close this issue for now.

closed 2016-07-22 by MichMich