MagicMirrorOrg/MagicMirror #377: Wrong temperature data shown in "Current Weather" Module

DirkBaumeister, 2016-07-20:
Hi there,

you have a little bug in your "Current Weather" module.

You use as weatherEndpoint "weather" which leads to a page like:

http://api.openweathermap.org/data/2.5/weather?id=[locationID]&units=metric&lang=de&APPID=[APPID]

But this page is different from this one:

http://api.openweathermap.org/data/2.5/weather**/**?id=[locationID]&units=metric&lang=de&APPID=[APPID]

There has to be a **/** at the end of the weatherEndpoint ;)

Best regards

MichMich, 2016-07-20:
It seems to be working. Are you experiencing any issues?

DirkBaumeister, 2016-07-20:
Compare:
http://api.openweathermap.org/data/2.5/weather?id=2810678&units=metric&lang=de&APPID=[APPID]

http://api.openweathermap.org/data/2.5/weather/?id=2810678&units=metric&lang=de&APPID=[APPID]

You see: Different Temperature in Main... With / is correct.

MichMich, 2016-07-20:
Fixed: https://github.com/MichMich/MagicMirror/commit/ae37fa2bc598301c69688c78a9adb0e270ad66ef

MichMich, 2016-07-20:
Honestly, I think Open Weather has some issues. I experienced the same yesterday. I do think the endpoint with the slash at the end looks better. SO I changed it as you said.

closed 2016-07-20 by MichMich

mbalfour, 2016-09-06:
The slash has been giving me some strange results ever since you added it - temperatures of ~302 degrees. :(  Without the slash I get the correct results.  (Basically, it's ignoring the units field and who knows what else)

According to the documentation, the correct calling syntax is "/weather?id=...".  See here:
http://openweathermap.org/current

The first example they list is this:  "api.openweathermap.org/data/2.5/weather?q={city name}"