MagicMirrorOrg/MagicMirror #388: Language default of 'en' in config.js results in a blank screen

causztic, 2016-08-01:
Hello,

I've experienced what other users have encountered over here https://forum.magicmirror.builders/topic/412/blank-black-screen-noob/18

Basically having the option of 'en' results in a blank screen. An empty string defaults it properly to English and works. Other language options works too.

MichMich, 2016-08-01:
Thanks for reporting. I'll investigate on the forum what is causing this issue. As soon as Its certain that this is a bug, I'll continue in this github issue.

MrHarcombe, 2016-08-01:
FWIW, I'm helping my son try and get this working (with a voice control module, by @dr4ke616) and we hit similar problems... but it's only for modules which use the translations part of the system - for us, initially, we just removed the default "alert" module and all seemed to be working (but any third party module that used the transations, caused the system to fail).

MichMich, 2016-08-01:
Interesting. I'll look into that.

MichMich, 2016-08-01:
@MrHarcombe is there a particular module you were using that caused the issues?

MrHarcombe, 2016-08-01:
As I say, the default "alert" module made things stop working. Having a default language of 'en' worked with the other default modules.... but I think "alert" is the only one that uses translations, isn't it?

MichMich, 2016-08-01:
I tried it with the alert module and the `en` language setting, and it seems to work fine here. Do you see any particular error in the developer tools console panel?

MrHarcombe, 2016-08-01:
The following config fails (browser [console log](https://github.com/MichMich/MagicMirror/files/394732/magicmirror-mrharcombe.c9users.io-1470083874259.txt) is attached...)
```
/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 */

var config = {
	port: 8080,

	language: 'en',
	timeFormat: 24,
	units: 'metric',

	modules: [
		{
			module: 'alert',
		},
		{
			module: 'clock',
			position: 'top_left',
			config: {
				displayType: 'analog',
				analogFace: 'face-001',
			}
		},
		{
			module: 'calendar',
			header: 'UK Holidays',
			position: 'top_right',
			config: {
				calendars: [
					{
						symbol: 'calendar-check-o ',
						url: 'webcal://www.calendarlabs.com/templates/ical/UK-Holidays.ics'
					}
				]
			}
		},
/*
		{
			module: 'currentweather',
			position: 'top_right',
			config: {
				location: 'New York',
				locationID: '',  //ID from bulk.openweather.org/sample/
				appid: 'YOUR_OPENWEATHER_API_KEY'
			}
		},
		{
			module: 'weatherforecast',
			position: 'top_right',
			header: 'Weather Forecast',
			config: {
	            location: 'New York',
				locationID: '5128581',  //ID from bulk.openweather.org/sample/
	            appid: 'YOUR_OPENWEATHER_API_KEY'
			}
		},
*/
		{
			module: 'newsfeed',
			position: 'bottom_bar',
			config: {
				feeds: [
					{
						title: "BBC UK News",
						url: "http://feeds.bbci.co.uk/news/uk/rss.xml"
					}
				],
				showSourceTitle: true,
				showPublishDate: true,
				showDescription: true,
			}
		},
	]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== 'undefined') {module.exports = config;}
```

MichMich, 2016-08-01:
That config seems to run fine here ... Don't see any errors in the log you've sent. 
This really is an odd issue. 

[image]

MrHarcombe, 2016-08-01:
That's a similar issue on both a RaspberryPi3 and also in my own c9.io workspace (which is running at the moment on http://magicmirror-mrharcombe.c9users.io:8080/ if you want me to tweak anything).

MichMich, 2016-08-01:
I will probably need to try it on a clean install on the a PI3 to see if I can get the same problem. Till then I can't really solve anything without a particular error message.

MrHarcombe, 2016-08-01:
Out of interest, I've put the config back to as it should be, which doesn't work for me; does it work for you?

MichMich, 2016-08-01:
Yes, if I open your url it works. If I copy your config, it works. If I use the default config, it works. I have no idea what the issue is ...

MrHarcombe, 2016-08-01:
Weird, as that currently doesn't work for me! Could it be down to what the system default language is (ie do you run your OS in English, by default?) and whether the config is pointing at a different language?

MichMich, 2016-08-01:
I run it on a english OS. But the way it's built it doesn't have anything to do with the OS. I suspect it has something to do with a dependency that gives some issues. Therefor I want to try a clean install, but unfortunately I currently don't have the time to try it out.

Are you using the master branch or the dev branch? And could you try "npm update" from the ~/MagicMirror folder?

MrHarcombe, 2016-08-01:
I'm using the tag v2.0.3 - but am happy to try any branch you need me to use. `npm update` didn't produce any output.

MichMich, 2016-08-01:
One issue could be that it can't find the translation file. Could you look in de dev tools if it somehow gives a 404 (not found) for one of the files it tries to load?

MrHarcombe, 2016-08-01:
No errors given in the browser console log - last entry is for the two translation files to be loaded, but no errors.

MrHarcombe, 2016-08-01:
Digging into the code, it doesn't seem to be outputting the log message in loader.js:153

MichMich, 2016-08-01:
Could you check the resources tab? (The console won't show a 404.)

MrHarcombe, 2016-08-01:
Um, I don't appear to have a resources tab (I'm in Chrome using the Inspector)?

MichMich, 2016-08-01:
It might be called the network tab.

MrHarcombe, 2016-08-01:
curl 'chrome-extension://pioclpoplcdbaefihamjohnefbikjilc/content/Clipper.js' -H 'Referer: http://magicmirror-mrharcombe.c9users.io:8080/' -H 'X-DevTools-Emulate-Network-Conditions-Client-Id: f285019f-74ab-4566-803a-f88e01577ad3,f285019f-74ab-4566-803a-f88e01577ad3,f285019f-74ab-4566-803a-f88e01577ad3,f285019f-74ab-4566-803a-f88e01577ad3,f285019f-74ab-4566-803a-f88e01577ad3,f285019f-74ab-4566-803a-f88e01577ad3' -H 'If-None-Match: "q1z7Go5Kn0fZ3tRatko3o/rSVTc="' -H 'Cache-Control: max-age=0' -H 'User-Agent: Mozilla/5.0 (X11; CrOS armv7l 8350.60.0) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/52.0.2743.85 Safari/537.36' --compressed

MrHarcombe, 2016-08-01:
No 404s, plenty of 304s (cache hits?)

MrHarcombe, 2016-08-01:
[image]

MichMich, 2016-08-01:
Hmmm no clue then. Will look at it asap.

MrHarcombe, 2016-08-01:
np - sorry I couldn't be more help :(

MrHarcombe, 2016-08-02:
Just to add to the confusion - the system works perfectly for me (including the voice control module) if I use a language of 'de' or 'fr'... frustrating, much?

FYI, the same symptoms appear if I comment out the "alert" module and just leave the "[MMM-Voice-Control](https://github.com/dr4ke616/MMM-Voice-Control)" module in, which uses translations; if the language is "en" then after the translation and fallbackTranslation are loaded, nothing else happens.

MrHarcombe, 2016-08-02:
Ooooh...

So, if I rearrange all the various places where translations are listed and make sure "en" isn't the fallback, then the system works for me just fine with a config.js language of "en" - it only seems to be the issue where the language and the fallback-language are the same.

Does that help any?

MichMich, 2016-08-02:
Sooooo ... just a few seconds ago I used a possible fix to the develop branch. Could you give it a try?

MichMich, 2016-08-02:
(and indeed, it was exactly what you said. fallback and language were pointing to same file, new electron refuses to reload file twice.)

MichMich, 2016-08-02:
If you want to give the develop branch a try:
```
cd ~/MagicMirror
git checkout --track origin/develop
```

MrHarcombe, 2016-08-02:
Works for me :)

Thank you 👍

MichMich, 2016-08-02:
Thanks for checking!

closed 2016-08-02 by MichMich

causztic, 2016-08-03:
The new electron-prebuilt v1.3.1 is also having issues with a phantom module version (It is using Node v6.3.0 which is ABI 48, but somehow in their repo they are built with ABI of 49). This might cause some issues with compilation for other custom modules - It affected my node-pocketsphinx implementation.

Check out https://github.com/electron-userland/electron-prebuilt/issues/172

MichMich, 2016-08-03:
@causztic, interesting. I'll keep an eye on that.

commit 33521ebd (2016-08-02): Fix #388