MagicMirrorOrg/MagicMirror #393: Blank screen

mohiuddinopu, 2016-08-07:
After following every step mentioned in the forum, app starts without any error, but the screen is blank.

stewartmckee, 2016-08-07:
I had the same issue and for me I used another machine and used chrome to open the console and found a javascript error in a 3rd party module.  You might find whats causing it in there?

MichMich, 2016-08-07:
The current master version contains a bug. Please try the develop branch to see if this fixes the issue for you.

stewartmckee, 2016-08-07:
FYI, I pulled yesterday evening from master and have it up and running in serveronly mode on a pi zero.  Not sure when the bug was introduced, but if after that, just ignore me. :)

MichMich, 2016-08-07:
Serveronly mode doesn't contain the bug. It's an electron issue. I'll release a new Master asap.

pi-mad, 2016-08-07:
How do you open develop branch

MichMich, 2016-08-07:
I just release a new master version. To update:

```
cd ~/MagicMirror
git pull
```

closed 2016-08-07 by MichMich

cyberdude86, 2017-12-16:
I am experiencing the same blank start issue with my freshly installed rpi3 with mm2.  Is there any current bugs in the master branch?