MagicMirrorOrg/MagicMirror #395: Calendar Module :  in x days give bad value

scarri, 2016-08-10:
Hi!
first, what a nice project this MagicMirror. 

I'm using the calendar module with two ical : facebook et garmin
On display, the module gives the right informations about event for today or tomorrow 

But for an event in 2 days or more, it seem calculate a wrong value...

For exemple (same problem with facebook and garmin calendar) : 
today is august 10. 
Birthday A : august 10 - shows today
Birthday B : august 11 - shows tomorrow
Birthday C : august 12 - shows in 1 day? Show display "in 2 days" 
Birthday D : august 14 - shows in 3 days? Show display "in 4 days" 

thanks 
Sébastien

MichMich, 2016-08-12:
It's the way moment.js does its calculations. If it is slightly less than 3 days (for example, 2 days and 23 hours) it will round it down to 2 days.

closed 2016-08-17 by MichMich

dgonano, 2016-11-21:
@MichMich, My calendar updates the days (3 days -> 2 days) at 12pm (midday) instead of 12am (midnight) for all-day events. Is it supposed to work like this?

MichMich, 2016-11-21:
@dgonano Honestly, I don't know ... :) You could check the moment.js documentation.

grigrichaminou, 2018-01-09:
Hello,

Thank you for your modules.

I have been trying for several days now to understand why there is a one-day delay in all my calendars.

For example, today is January 9,2018:

event on January 10,2018 so in 1 day --> today at 8:00 am
event on January 11,2018 so in 2 days --> tomorrow at 8:00 am

I don't understand I have read a lot and tried it out but nothing works. My time zone is well configured.

Do you know the reason ?

best regards