MagicMirrorOrg/MagicMirror #396: How do I access the config/config.js on my raspberry pi 3?

vbhatti90, 2016-08-12:
I have installed the MagicMirror2 program. Now i want to add the config/config.js.sample text to config/config.js. How do i access config/config.js on the raspberry pi 3?

What commands should  be using?

jesussanchez-a, 2016-08-12:
[image]
[image]

Follow the commands:
ls
cd ~/MagicMirror
ls
cd config
ls
sudo cp config.js.(prefix if any) config.js
sudo nano config.js

MichMich, 2016-08-12:
For future support request please use http://forum.magicmirror.builders

closed 2016-08-12 by MichMich