MagicMirrorOrg/MagicMirror #402: Autostart issues

Lunamick, 2016-08-16:
Hello all

Sorry im new to this and have been self teaching myself for the past few weeks.

I got everything up and running> I just cant seem to get the autostart function to work.

any help or walkthrough?

i followed the directions precisely and have restarted from scratch 4 times and still cant get it to work.

Thanks

pcp47, 2016-08-17:
Habe you made this Step?

Install PM2

Install PM2 using NPM:

sudo npm install -g pm2
Starting PM2 on Boot

To make sure PM2 can do it's job when (re)booting your operating system, it needs to be started on boot. Luckily, PM2 has a handy helper for this.

pm2 startup
PM2 will now show you a command you need to execute.

Make a MagicMirror start script.

To use PM2 in combination with MagicMirror, we need to make a simple shell script. Preferable, we put this script outside the MagicMirror folder to make sure it won't give us any issues if we want to upgrade the mirror.

cd ~
nano mm.sh
Add the following lines:

cd ~/MagicMirror
DISPLAY=:0 npm start
Save and close, using the commands CTRL-O and CTRL-X. Now make sure the shell script is executable bij performing the following command:

chmod +x mm.sh
You are now ready to the MagicMirror using this script using PM2.

Starting your MagicMirror with PM2

Simply start your mirror with the following command:

pm2 start mm.sh
You mirror should now boot up and appear on your screen after a few seconds.

Enable restarting of the MagicMirror script.

To make sure the MagicMirror restarts after rebooting, you need to save the current state of all scripts running via PM2. To do this, execute the following command

pm2 save

MichMich, 2016-08-17:
I close this issue for now. If you have additional questions, please open a support post on http://forum.magicmirror.builders

closed 2016-08-17 by MichMich