MagicMirrorOrg/MagicMirror #404: Newsfeed suspend

barnabycolby, 2016-08-18:
When the newsfeed is hidden, the stories continue to show, causing all stories shown during that time to never be shown. An implementation of the suspend/resume functions could be added to pause the update timers.

MichMich, 2016-08-22:
The news items will be repeated until their outdated anyway. So I don't see the real issue here?

barnabycolby, 2016-08-26:
This is definitely a minor edge case, but consider the following example:
- There are 5 news stories in the feed, each displayed for 10 seconds
- Another plugin (such as [MMM-Carousel)](https://github.com/barnabycolby/MMM-Carousel) hides the news feed plugin every 40 seconds for a duration of 10 seconds
- In this case, the fifth story would never be displayed

MichMich, 2016-08-26:
Ah yes. I understand. Feel free to send an PR. Oherwise I'll look into it when I have some spare time.

deadherring, 2017-01-12:
@MichMich please explain your comment "the news item will be repeated until they are outdated anyway".what makes it outdated? I am finding that news items stay (so far up to 20 days) and the news feeds do not update.  Please explain if I am doing something wrong, I can't get answers in the forum. 

Thanks!

Nathan

MichMich, 2017-01-15:
It will just repeat the items in the RSS feed. So if the RSS feed if updated (with old stories removed), they won't show up anymore.

deadherring, 2017-01-17:
@MichMich Sorry if I'm being dense, but I'm not following.  Right now my MM news feed does not update.  It is showing news headlines from 20+ days ago.  If I reboot the pi it will update the headlines for a few days and then display old headlines going forward.

Is that expected behavior?  A bug?  Something in the way I've configured it?

Thanks,

Nathan

MichMich, 2017-01-17:
Maybe this helps for you? https://forum.magicmirror.builders/topic/1429/mmm-newsfeed

deadherring, 2017-01-17:
@MichMich Cool, I'll give that a try.  BTW, thanks so much for developing and releasing the software.  Everyone who sees my MM goes googly eyed.  Somehow my wife agreed to let me cut a hole in the bathroom wall and install it so it's flush with the wall and it came out great (I have a woodworking shop in my basement so I was set on the woodworking side).  Here's how it came out, if you are interested: http://imgur.com/a/n63NT

Also, I was at the Consumer Electronics show in Las Vegas earlier this month and I saw a company had a version of the magic mirror they were showing.  I can't remember the name of the company, though.

Thanks,

Nathan

MichMich, 2017-01-18:
@deadherring Could you talk to my wife about cutting a hole in our wall? ;)

closed 2017-01-18 by MichMich