MagicMirrorOrg/MagicMirror #411: use npm as module manager

talarari, 2016-08-22:
you can use npm as the module manager
that way installing a module will just be installing an npm module.

that way modules can be versioned and easier to install and manage.

MichMich, 2016-08-22:
Unfortunately there is no way with NPM to say where the module must be placed. They'll always be placed in the node_modules folder. This would require a rewrite.

talarari, 2016-09-05:
it would require some code changes, but why a rewrite?
one nice way to use npm as the module repository is to use a prefix for magic mirror modules.
(like they do in this project: https://github.com/appetizermonster/hain)
so modules will be something like magic-mirror-stocks

that way you can look for modules in both modues and node_modules (for backwards compatibility)

paviro, 2016-09-05:
Had exactly the same thought on my way to work today! :D

MichMich, 2016-09-05:
It hasnt got my priority yet. Feel free to send a PR.

nhubbard, 2016-09-14:
@MichMich: I would love to look at this issue, but I can't find which lines directly reference the modules folder in `js/loader.js`. If I get some line numbers, I can figure out if this is a feasible idea.

MichMich, 2016-09-15:
It's not just a few lines you need to change. A lot it tight to the folder of a module. (Like a web server for hosting additional files ...) But a good place to start is https://github.com/MichMich/MagicMirror/blob/develop/js/loader.js#L86

MichMich, 2017-01-19:
I don't think this is going anywhere soon. Fee free to reopen if necessary.

closed 2017-01-19 by MichMich

EliteByte, 2017-07-01:
Hello, sorry for bumping this thread after so long but I am currently trying to develop a solution to this problem using a Material (material.io) themed website in which people can download a package to their Raspberry Pi and run probably a node server to update and configure their modules. 
[image] I was wondering if, as of recently their has been any applications like it before I go any further.

MichMich, 2017-07-01:
My Best guess is to ask it on the forum. There were some users working on it.

fewieden, 2017-07-01:
https://github.com/Jopyth/MMM-Remote-Control