MagicMirrorOrg/MagicMirror #412: Is there a way for calendar to show 24 hour times?

dinkybluebug, 2016-08-22:
Is there anyway for calendar to show 24 hours times?  as in the clock?

Many thanks.

MichMich, 2016-08-23:
Which Locale (Language) are you using?

dinkybluebug, 2016-08-23:
Thanks for replying.

My config.js says:

	language: 'en',
	timeFormat: 24,
	units: 'metric',

Is this what you mean?

dinkybluebug, 2016-08-24:
@MichMich Is this the information you were after?

MichMich, 2016-08-24:
Yeah, sorry for the slow response, I'm currently at a developers conference. Anyway, the way the time is formatted is done by moment.js. Moment formats the date/time based on the locale. You could look into the moment.js documentation if there is an other locale that has english text but a 24 hour formatting.

I agree, this is not the best route and a modification to the calendar would be a better option.

dinkybluebug, 2016-08-24:
@MichMich No worries , enjoy your conference.  I'm a bit of a newby when it comes to all of this - but I'm learning lots.  I'll have a look and the moment.js documentation and post what I find.  Thanks for the response.

Marc

dinkybluebug, 2016-08-24:
apparently changing the moment.js locale to 'en-gb' does this  - it reverts to 24 hr clock rather than US english - 'en'  

Any advice about what needs changing to reflect this?

MichMich, 2016-08-24:
You only need to update your language field in the config. The value you fill in here, will be directly send to the moment.js instance.

dinkybluebug, 2016-08-24:
it works it works.  hurrah!

Thanks!

closed 2016-08-24 by dinkybluebug