MagicMirrorOrg/MagicMirror #424: Adding Backstroke support?

nhubbard, 2016-08-31:
I noticed on [github/linguist](https://github.com/github/linguist) that they use a service called Backstroke to keep forks up to date and prevent merge conflicts.

Here is a link to it: [1egoman/backstroke](https://github.com/1egoman/backstroke)

Should we enable this service? I think it would help, because I constantly fight with Git over merging new commits that I haven't pulled down yet.

MichMich, 2016-08-31:
I'm not really seeing how this would improve the workflow since we have a develop branch. Just always do a git pull before you start working. I think it would just overcomplicate stuff.

nhubbard, 2016-08-31:
It helps out, especially when you have a fork that you use to host the Magic Mirror. (I have a private fork that I use to host my personal mirror on Windows Azure Websites.) It just makes a pull request that I can merge to update my fork. It also makes it easier for people (like me) who need to use many different.

MichMich, 2016-08-31:
There are already a lot of people having issues with the git workflow (pull requests etc), since MagicMirror is a somewhat "Starter project" for many of them. Implementing something like this would increase the threshold.

Maybe reconsider the use of a private fork? I don't see the use of that. I have many different mirrors running, all of them just use the master or develop branch. If that's not possible you're probably an edge case.

nhubbard, 2016-08-31:
OK, I understand your concern.

closed 2016-08-31 by nhubbard