MagicMirrorOrg/MagicMirror #438: Calendar not loading

pagoa, 2016-09-12:
Hi!

First of all: Great job and thanks for investing your time and sharing your work with everybody.

BUT I have several problems. I have only tried to run the server and access it to configure it, prior to installing the mirror. My problem is that the calendar, as well as the news feeds are not working (also the additional email module...) AND i cannot find any feedback whatsoever. Is there supposed to be a logfile somewhere? All I get is a Warning in the console:

WARNING! Could not load config file. Starting with default configuration. Error found: TypeError: Object #<Object> has no method 'accessSync'

The weather works just fine. When I try to run it with dev option to get more feedback, it refuses to launch and drops following log:

0 info it worked if it ends with ok
1 verbose cli [ 'node', '/usr/bin/npm', 'start', 'dev' ]
2 info using npm@3.10.7
3 info using node@v0.10.41
4 verbose node symlink /usr/bin/node
5 verbose run-script [ 'prestart', 'start', 'poststart' ]
6 info lifecycle magicmirror@2.0.0~prestart: magicmirror@2.0.0
7 silly lifecycle magicmirror@2.0.0~prestart: no script for prestart, continuing
8 info lifecycle magicmirror@2.0.0~start: magicmirror@2.0.0
9 verbose lifecycle magicmirror@2.0.0~start: unsafe-perm in lifecycle true
10 verbose lifecycle magicmirror@2.0.0~start: PATH: /usr/lib/node_modules/npm/bin/node-gyp-bin:/home/pi/MagicMirrorBU/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games:/usr/local/lib/:/usr/local/bin
11 verbose lifecycle magicmirror@2.0.0~start: CWD: /home/pi/MagicMirrorBU
12 silly lifecycle magicmirror@2.0.0~start: Args: [ '-c', 'electron js/electron.js "dev"' ]
13 silly lifecycle magicmirror@2.0.0~start: Returned: code: 1  signal: null
14 info lifecycle magicmirror@2.0.0~start: Failed to exec start script
15 verbose stack Error: magicmirror@2.0.0 start: `electron js/electron.js "dev"`
15 verbose stack Exit status 1
15 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/lib/utils/lifecycle.js:255:16)
15 verbose stack     at EventEmitter.emit (events.js:98:17)
15 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/lib/utils/spawn.js:40:14)
15 verbose stack     at ChildProcess.emit (events.js:98:17)
15 verbose stack     at maybeClose (child_process.js:766:16)
15 verbose stack     at Process.ChildProcess._handle.onexit (child_process.js:833:5)
16 verbose pkgid magicmirror@2.0.0
17 verbose cwd /home/pi/MagicMirrorBU
18 error Linux 3.18.7-v7+
19 error argv "node" "/usr/bin/npm" "start" "dev"
20 error node v0.10.41
21 error npm  v3.10.7
22 error code ELIFECYCLE
23 error magicmirror@2.0.0 start: `electron js/electron.js "dev"`
23 error Exit status 1
24 error Failed at the magicmirror@2.0.0 start script 'electron js/electron.js "dev"'.
24 error Make sure you have the latest version of node.js and npm installed.
24 error If you do, this is most likely a problem with the magicmirror package,
24 error not with npm itself.
24 error Tell the author that this fails on your system:
24 error     electron js/electron.js "dev"
24 error You can get information on how to open an issue for this project with:
24 error     npm bugs magicmirror
24 error Or if that isn't available, you can get their info via:
24 error     npm owner ls magicmirror
24 error There is likely additional logging output above.
25 verbose exit [ 1, true ]

I'm running it on a raspberry pi 2 on jessie. The config file should be fine and I have already tried to retrieve the calendar data with the curl command (curl -sL URL_TO_CALENDAR | bash) => looking good, I can see the events there.
Any ideas on how to fix it would be appreciated :)

MichMich, 2016-09-12:
For support, please check the forum: http://forum.magicmirror.builders

closed 2016-09-12 by MichMich

pagoa, 2016-09-14:
Waiting for an email to finish registration at "http://forum.magicmirror.builders" ever since unfortunately. I've already tried two different email adresses; no luck so far.

Anyway solved the problem in a weird way myself now:
Removed node and npm using "apt-get remove --purge" because I thought I might give reinstallation a try. However I must have had several versions installed because I executed "node serveronly" (to check whether it has actually been removed) and it suddenly all worked just fine o.O (as long as I don't use the epiphani browser which only shows a blank white page) ...

paviro, 2016-09-14:
Can you trie again we changed some stuff with our mail provider :)

pagoa, 2016-09-15:
> Can you trie again we changed some stuff with our mail provider :)
Yap working now; thx :)