MagicMirrorOrg/MagicMirror #440: sendSocketNotification only works from module to node_helper

atainter, 2016-09-14:
I'm trying to develop my own module. `this.sendSocketNotification` only sends a message from the module file to a node_helper. For some reason, I can't receive a socket notification in my module from the node_helper.

Here is my module code:

```javascript
Module.register("helloworld",{

	// Default module config.
	defaults: {
		text: "Hello World!"
	},

	// Define start sequence.
	start: function() {
		this.sendSocketNotification("ADD_FEED", {"test":"test"})
	},

	// Override dom generator.
	getDom: function() {
		var wrapper = document.createElement("div");
		wrapper.innerHTML = this.config.text;
		return wrapper;
	},

	// Override socket notification handler.
	socketNotificationReceived: function(notification, payload) {
		console.log("module received: " + notification)
	}
});
```

Here is my node_helper:
```javascript
var NodeHelper = require("node_helper");

module.exports = NodeHelper.create({
	// Subclass start method.
	start: function() {
	},

	// Subclass socketNotificationReceived received.
	socketNotificationReceived: function(notification, payload) {
		console.log("helper received: " + notification)
		this.helperFunc()
	},

	helperFunc: function() {
		this.sendSocketNotification("TEST", {"test":"test"})
	}
})
```

...and my config file:

```javascript
var config = {
	port: 8080,

	language: 'en',
	timeFormat: 24,
	units: 'metric',

	modules: [
    {
        module: 'helloworld',
        position: 'bottom_bar', // This can be any of the regions.
        config: {
            // See 'Configuration options' for more information.
            text: 'Hello world!'
        }
    }
	]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== 'undefined') {module.exports = config;}
```

MichMich, 2016-09-14:
It works both ways, but the client side must establish the connection to the node helper first. So you need to send a message from the client to the node helper to begin with.

atainter, 2016-09-14:
From helloworld.js:
```javascript
// Define start sequence.
    start: function() {
        this.sendSocketNotification("ADD_FEED", {"test":"test"})
    },
```

I have ☝️

MichMich, 2016-09-14:
Then I suggest to open a topic on http://forum.magicmirror.builders, which is the official location to get support. 👍

atainter, 2016-09-14:
That works. I see `helper received: ADD_FEED` in the console.

atainter, 2016-09-14:
It looks like there is a two way connection, but for some reason I cannot log from the module.

MichMich, 2016-09-14:
Where are you looking for your log? Cause the node helper doesn't log in your browser window.

atainter, 2016-09-14:
I was seeing terminal logs for node_helper, but not the module. I've never used electron before, so I was unaware there is a console like the browser. I see them now in the electron console. Thanks!

closed 2016-09-14 by atainter

priya99, 2017-05-19:
Hi,

I am new to node.js.I am trying to create new sample module "test".I am sending notification from test module to node_helper and trying to receive notification back to test module.From test module to node_helper notification works fine, but test module does not receive any notification from node_helper.

Any help appreciated
Here is my module code:
test.js:

Module.register("test",{

	defaults:{
		text:"Happy Birthday!!!!"
	},
	
	getDom: function() {
		var wrapper = document.createElement("div");
		wrapper.innerHTML =this.config.text;
		return wrapper;
	},
	
	start:function(){
		this.sendSocketNotification("main",{message:"test1"});
	},
	
	
	socketNotificationReceived: function(notification, payload) {
		console.log('socketNotificationReceived() in main module..'+notification);
		if (notification === "test2") {
			console.log('test2 socket notification received...');
		}
	}
	
});

Here is node_helper.js:

var NodeHelper = require("node_helper");

module.exports = NodeHelper.create({
	// Override start method.
	start: function() {
		
	},

	// Override socketNotificationReceived method.
	socketNotificationReceived: function(notification, payload) {
		var self=this;
		console.log('*Starting socketNotificationReceived()*'+notification);
		if (notification === "main") {
			console.log(notification+' socket notification received...');
			this.function1();
		}
	},
	
	function1:function(){
		console.log('inside function1');
		this.sendSocketNotification("test2",{message:"test2"});
	}
});
	

Here is config.js:

var config = {
	port: 8080,
	ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"], // Set [] to allow all IP addresses.

	language: "en",
	timeFormat: 24,
	units: "metric",

	modules: [
		
		{
			module: "test",
			position: "top_right"
		}
        ]
   };

saraalvin, 2017-05-30:
Same problem as @priya99 here!

roramirez, 2017-05-31:
Here are a module create example where you can use the notices in both ways

https://github.com/roramirez/MagicMirror-Module-Template

fewieden, 2017-05-31:
@saraalvin he didn't had an issue with the code, it was a problem of understanding how the system works see #894

saraalvin, 2017-05-31:
Hi! Thanks for the quick reply :)

I think I'm following the general rules: 
- I send a notification from module to node_helper in the start function.
- I understand that module and node_helper logs appear in different consoles.
- As suggested [here](https://github.com/MichMich/MagicMirror/issues/196), I avoid calling directly to sendSocketNotification from start() in the module, using an intermediate function to do it.

My problem is that I do receive the notification from node_helper to the module, but not the content of the payload. This is, if I send as payload, for example, a string from node_helper, the module receives 'undefined'. 

This is the code in node_helper that sends the notification:
```
socketNotificationReceived: function(notification, payload){
    if (notification === "CONNECT"){

      // Do things to get exec, fs, mic, ... objects

      var variables = {
        exec: exec, 
        fs: fs, 
        mic: mic, 
        speech_to_text: speech_to_text, 
        text_to_speech: text_to_speech, 
        conversation: conversation
      };

      this.sendSocketNotification("REQUIRED_VARIABLES", variables);
}
```
And this is the part in the module that receives it:

```
    socketNotificationReceived: function(notification, payload){
        if (notification === "REQUIRED_VARIABLES"){
            this.myFunction(payload);
        }
    }
```
Maybe I'm missing something?
Thx!