MagicMirrorOrg/MagicMirror #442: New feature (for developers?): Update header with optional information

Jopyth, 2016-09-16:
I like headers for my modules, buy at the same time the need additional space. Therefore I think it is useful to be able to add tiny bits of information to the header, which otherwise would need extra space in the module content.

An implementation example exists on my [fork](https://github.com/Jopyth/MagicMirror). I could create a clean pull request, with only the modifications regarding this `appendix` to the header (open for other name suggestions ;) ).

For example we can show, when the last data was pulled in the backend, or how many contacts from the address book have been loaded, small hints if an error occurs, etc.
[image]

Maybe related losely to #236.

MichMich, 2016-09-17:
Yeah, this is something on my wishlist/todo list. Will probably add this feature next month.

MichMich, 2016-09-17:
Of course a PR is welcome, but it depends on the implementation. I'm a little picky for the core changes.

Jopyth, 2016-09-17:
Well then start picking on #444 when you have time. Or you can just use this as a basis for your ideas.

MichMich, 2016-09-20:
I've added this functionality. See: https://github.com/MichMich/MagicMirror/commit/b2a7d3584bad2752d59d2325c581f5018e875306

Example code:
````
getHeader: function() {
    return this.data.header + ' appendix';
}
````

Let me know if this works for you.

Jopyth, 2016-09-20:
Yes, that looks perfect. :)

Jopyth, 2016-09-20:
Example:

+````javascript
+getHeader: function() {
+	return this.data.header + ' <span class="fa fa-refresh"></span> ' + this.lastUpdated;
+}
+
+````

This one did not work for me inside a `header` element. However it worked inside a `div`.

Although I am wondering why [this post](http://stackoverflow.com/a/19779520), says that for example `span` elements should work. Maybe there was a different error. Can you confirm, that it is possible to include a font-awesome icon in the header? That would be awesome.

MichMich, 2016-09-20:
Not sure. It should return a string. What error do you experience?

MichMich, 2016-09-20:
If this really is an issue, I wouldn't mind moving to a dig container in stead of a header. Feel free to send a PR for this. Otherwise I can make an adjustment somewhere later this week.

Jopyth, 2016-09-21:
Never mind, my bad. I guess there must have been some other error. The `span` tag works just fine inside the `header` tag.

When I originally implemented this, it appeared as if it did not resolve the html `span` tag correctly (displaying the raw code instead of the font awesome symbol). But it works perfectly fine now.

MichMich, 2016-09-21:
Good to hear. Looking forward to your module and see how you implemented it.

closed 2016-09-21 by MichMich