MagicMirrorOrg/MagicMirror #443: New feature: Alternative way of limiting space of modules exactly

Jopyth, 2016-09-16:
I was curious about the calendar module, as it limits the title length to a certain number of characters. Now I thought, maybe we can limit this by pixels instead, as title length varies based on the text. Also what can I do to read the remaining text, while still limiting the total width to a reasonable size.

So these scrollings text effects came to my mind, and based on this (StackOverflow post)[http://stackoverflow.com/questions/30671966/create-scrolling-text-when-text-is-too-long] I implemented something similar to (this)[http://css3.bradshawenterprises.com/transitions/]. Is this worth a pull request? (Read as: Does somebody else have a tiny tiny screen behind their mirror?)

Basically I can optionally specify a number of pixels for a text `span` element, and if the text doesn't match the overflow is hidden, but scrolled back and forth every few seconds.

MichMich, 2016-09-17:
It won't have my highest priority. But feel free to send a PR.

MichMich, 2017-01-19:
I don't think this is going anywhere soon. Fee free to reopen if necessary.

closed 2017-01-19 by MichMich