MagicMirrorOrg/MagicMirror #456: Limit access to localhost (security related)

Jopyth, 2016-09-28:
Are there any merits to the usual workflow, if we limit the access to localhost? For example change [this line](https://github.com/MichMich/MagicMirror/blob/master/js/server.js#L17) to 
`server.listen(config.port, config.route);` and make the default `route = "127.0.0.1"`.

Reason: More and more modules do very cool stuff (e.g. accessing Emails), however they also need the login information in the config. Currently anyone on your network (guests - if you do not have a separate guest network) can straight up load your mirror html page or even your `config.js` and read sensitive information from your config file.

So if I invite Eve to my house and she goes to 192.168.0.100:8080/config/config.js (if that is my Raspberry Pi IP) she gets the good old `Server not responding` or whatever instead of reading all our API keys, possible usernames and password for certain websites, etc...

There is probably a much better way, but for the moment this would be very easy to implement.

MichMich, 2016-09-28:
You make very valid point. But I think it might be a good idea to make this configurable. Since a lot of people access it from a external IP to test it. Maybe we should add whitelist IP's. 

Most important: never allow people you don't trust on your network! 😉

Jopyth, 2016-09-28:
Exactly that was my thought behind the `config.route` value. We would set this in the config file, so just like changing the port we can change the route. If this is set to "0.0.0.0" everything is like it is now, everyone can access it.

Something like a Whitelist would be awesome.

Still I think, to protect the unsuspecting user making the default (e.g. In the config.js.sample) to only allow access from localhost would be good. Plus of course documenting this change for anyone who is developing.

MichMich, 2016-09-29:
Will look into this if I have time. Feel free to send a PR.

Jopyth, 2016-09-29:
Would you prefer the simple all or nothing approach, or a more comprehensive one, for example with [express-ipfilter](https://www.npmjs.com/package/express-ipfilter)? With this package we can use an `ip_whitelist`, but it adds another dependency.

MichMich, 2016-09-29:
I would prefer the latter. MagicMirror² relies on a lot of dependencies already so that's not really an issue.

closed 2016-09-29 by Jopyth