MagicMirrorOrg/MagicMirror #458: Current Weather Module Not Updating

moviemanic, 2016-09-29:
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Raspberry Pi 2

**MagicMirror Version**: JavaScript version (v2).

**Description**: All of my other modules (Clock, WUnderground, Calendar, News Feed) are updating just fine however my Current Weather module will only update when MagicMirror is first launched, from then on the temperature remains the same even though open weather shows a different temp.

**Steps to Reproduce**: I did not make any changes to the Current Weather module. 

**Expected Results**: The correct temp displaying, with updating every 10 minutes.

**Actual Results**: The temp updates on launch and never again even though internet connection is maintained and my other modules update.

**Configuration**: The configuration for the Current Weather module is:
{
        module: 'currentweather',
        position: 'top_right',  // This can be any of the regions.
                                    // Best results in left or right regions.
        config: {
            // See 'Configuration options' for more information.
            location: 'location,location',
            locationID: 'ID', 
            appid: 'abcde12345abcde12345abcde12345ab' 
        }
    }
]

MichMich, 2016-09-29:
Do you see any error or messages in your browser's console?

MichMich, 2016-10-01:
For further support, please visit http://forum.magicmirror.builders.

closed 2016-10-01 by MichMich