MagicMirrorOrg/MagicMirror #465: There seems to be an issue with the Kiosk mode and going to fullscreen with Electron.

neelrahul, 2016-10-05:
**Platform**: Raspberry Pi 3

**Node Version**: node --version  v6.7.0

**MagicMirror Version**: newer JavaScript version (v2).

**Description**:  Not getting full screen in "kioskmode" Instead I am getting the toolbar.

**Steps to Reproduce**: I set kioskmode: 'true' in the config.js. but electron would not set the window to kiosk mode, I still get the tool bar on the top and have to manually click on the toolbar and choose the toggle full screen option.

When I remove the kioskmode: 'true' then it went to the fullscreen mode without the toolbar like a true kiosk.

Here are the lines in the electron.js. It appears that when you choose kioskmode, for some reason it is not honoring the kiosk:true. By not choosing kioskmode, its going to the fullscreen and hiding the menu bar, which is the desired look while in a "kiosk" mode.

function createWindow() {
        // Create the browser window.
        if (config.kioskmode) {
                mainWindow = new BrowserWindow({width: 800, height: 600, x: 0, y: 0, kiosk:true, darkTheme: true, webPreferences: {nodeIntegration: false}});
        } else {
                mainWindow = new BrowserWindow({width: 800, height: 600, x: 0, y: 0, fullscreen: true, autoHideMenuBar: true, darkTheme: true, webPreferences: {nodeIntegration: false}});
        }

**Expected Results**: Full screen with no toolbar.

**Actual Results**: Full screen with toolbar

**Configuration**:  Note: I have commented out the kiosmode line to get my fullscreen with no toolbar.

/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 */

var config = {
	port: 8080,

	language: 'en',
//	kioskmode: 'true',
	timeFormat: 12,
	units: 'imperial',

	modules: [
		{
			module: 'alert',
		},
		{
			module: 'clock',
			position: 'top_left',   // This can be any of the regions.
        		config: {
			showPeriodUpper: 'true',
			clockBold: 'true',
			analogFace: 'face-002',
			displayType: 'both'
        			}
		},
//		{
//			module: 'calendar',
//			header: 'US Holidays',
//			position: 'top_left',
//			config: {
//				calendars: [
//		{
//						symbol: 'calendar-check-o ',
//						url: 'webcal://www.calendarlabs.com/templates/ical/US-Holidays.ics'
//					}
//				]
//			}
//		},
		{
			module: 'compliments',
			position: 'lower_third'
		},
		{
			module: 'currentweather',
			position: 'top_right',
			config: {
				location: '',
				locationID: '0000000',  //ID from http://www.openweathermap.org
				appid: 'sdfsdfsdfsdfsdf'
			}
		},
		{
			module: 'weatherforecast',
			position: 'top_right',
			header: 'Weather Forecast',
			config: {
	            location: ',
				locationID: '00000000',  //ID from http://www.openweathermap.org
	            appid: 'sdfsdfsfsfsffs'
			}
		},
		{
			module: 'newsfeed',
			position: 'bottom_bar',
			config: {
				feeds: [
					{
                                		title: "BBC",
                                		url: "http://feeds.bbci.co.uk/news/rss.xml?edition=int",
					},
					{
						title: "New York Times",
						url: "http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml"
                        }
				],
				showSourceTitle: true,
				showPublishDate: true
			}
		},
	]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== 'undefined') {module.exports = config;}

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

MichMich, 2016-10-05:
The kiosk mode is not necessary when you're running on the PI. Is there a special reason why you're running in Kiosk Mode?

neelrahul, 2016-10-05:
Probably my lack of knowledge :). I was under the assumption that Kiosk Mode was if I didnt want any windows to popup on top of the MM screen.

closed 2016-10-05 by neelrahul