MagicMirrorOrg/MagicMirror #466: Default calendar with https and self signed certificates

kodejak, 2016-10-09:
Hello,
I'm new to MagicMirror and tried to get my calendar of a self hosted owncloud / nextcloud instance with self signed certificate.
It doesn't work out-of-the-box. So I decided to test this.

I had to change three files to get this work. Additional to pass a new calendar config entry from `calendars: []` to the call chain (parameters), I had to edit the file "calendarfetcher.js":

After line 28++:
		var opts = {
			headers: {
				'User-Agent': 'Mozilla/5.0 (Node.js 6.0.0) MagicMirror/v2 (https://github.com/MichMich/MagicMirror/)'
			}
};

I added:
```
if (selfSignedCert) {
                    var agentOptions = {
                        rejectUnauthorized: false
                    };
                    opts.agentOptions = agentOptions;
                }
```
This work as well. But I don't know any side effects of this!

Please make self signed certs to work with calendar app.

And forgive this submitting style, because I have no idea how to push code without cloning your repo to my account. I'm very new to git :)

MichMich, 2016-10-11:
Since this is an edge case, I will probably won't change this in the near future. Feel free to send a well documented pull request.

StefanBols, 2016-11-15:
Hi @kodejak 

You can't push code directly to this repo, without having developer privileges. Just fork the repo and make your changes and then submit a pull request:-)
If you want a quick and simple crash course, just say the word.

qistoph, 2016-12-29:
Accepting self-signed certificates blindly poses a security risk. It would be like not using HTTPS at all and any data transmitted over the connection can be manipulated or intercepted.

I would suggest you to try and get a proper certificate. Have a look at [Let's Encrypt](https://letsencrypt.org/) for a free certificate that is properly signed and authenticated.

If MichMich thinks the way forward for MagicMirror would be to allow self-signed certificates, I would opt to have the list of accepted certificates to be configurable and absolutely not blindly accepting any self-signed certificates!

MichMich, 2017-01-15:
Since I don't think anyone will work on this, I close this issue for now. Feel free to reopen.

closed 2017-01-15 by MichMich

codac, 2021-02-13:
@kodejak : could you please provide the rest of the source code, then I can change the code and add a pull request.
It's driving me crazy that my personal calendar is not working just because I'm using a self-signed cert....

qistoph, 2021-02-13:
I hope the PR will include more than just the current code change.

This one, as stated before, makes MagicMirror accept any certificate and thus effectively disables any https security. 

Please at least make this a configurable option instead of default behavior.

In the last couple of years software and browser companies have been trying to work on a safer web, also based on https. Let's try and keep all software as safe as possible and not just blindly accept any certificates without warning.

codac, 2021-04-14:
Although testing was successfull, with the new release 2.15.0 of MagicMirror, even with the "selfSignedCert" Switch set in the config.js I now get the error "UNABLE_TO_VERIFY_LEAF_SIGNATURE". It seems that setting the User Agent Option "rejectUnauthorized: false" doesn't work. Maybe it has something to do with the new change in this version "Removed ical which was substituted by node-ical in release v2.13.0. Module developers must install this dependency themselves in the module folder if needed." ?

It is quite hard to debug, can anyone help and support to find the error?

Furthermore I would like to add this parameter to the documentation of the calendar module:
https://docs.magicmirror.builders/modules/calendar.html

What do I need to do in order to add the parameter?

khassel, 2021-04-14:
I have no setup to test this. In `modules/default/calendar/calendarfetcher.js` is already this part included

````
			httpsAgent = new https.Agent({
				rejectUnauthorized: false
			});
````

You can add another line and test if this works

````
			httpsAgent = new https.Agent({
				strictSSL: false,
				rejectUnauthorized: false
			});
````

By the way these options are disabling (more or less) ssl, so I don't see any advantage in using that over http.

codac, 2021-04-15:
It seems that the above mentioned code wasnt the problem.
Adding strictSSL: false didn't help.

There was a wrong option "httpsAgent" used in the fetch instead of the correct option "agent".

nufan66, 2021-11-25:
I still get this error. So where do I have to change "httpsAgent" to "agent"?