MagicMirrorOrg/MagicMirror #472: Error: Invalid UNTIL value: 20160524T000000

thale13, 2016-10-17:
Not sure if this is just me or something else, but I am trying to connect a MagicMirror to my Cozi calendar (cozi.com) and I get that error with a Java stack trace.

Whoops! There was an uncaught exception...
Error: Invalid UNTIL value: 20160524T000000
    at Object.untilStringToDate (/home/pi/MagicMirror/node_modules/rrule/lib/rrule.js:210:19)
    at Function.RRule.parseString (/home/pi/MagicMirror/node_modules/rrule/lib/rrule.js:1453:42)
    at Function.RRule.fromString (/home/pi/MagicMirror/node_modules/rrule/lib/rrule.js:1467:28)
    at Object.ical.objectHandlers.END (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/node-ical.js:42:24)
/* Magic Mirror Config Sample
    at Object.handleObject (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/ical.js:257:41)
    at Object.parseICS (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/ical.js:298:20)
    at Request._callback (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/node-ical.js:11:24)
    at Request.self.callback (/home/pi/MagicMirror/node_modules/request/request.js:187:22)
    at emitTwo (events.js:106:13)
    at Request.emit (events.js:191:7)
MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
^C

Platform: RasPi 3 on 4.1.18-v7+ #846 SMP Thu Feb 25 14:22:53 GMT 2016 armv7l GNU/Linux

**Node Version**: Make sure it's version 0.12.13 or later.
node v6.7.0

**MagicMirror Version**: Now that the versions have split, tell us if you are using the PHP version (v1) or the newer JavaScript version (v2).
It's gotta be v2 as I just downloaded it last week.

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.
trying to connect to a published (to me) ics url

**Steps to Reproduce**: List the step by step process to reproduce the issue.
start magicmirror after configuring the calendar

**Expected Results**: Describe what you expected to see.
See events

**Actual Results**: Describe what you actually saw.
See no events

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!
               {
                        module: 'calendar',
                        header: 'Calendar',
                        position: 'top_left',
                        config: {
                                calendars: [
                                        {
                                                symbol: 'calendar',
                                                url: http://rest.cozi.com/api/.../..../feed.ics'
                                        }
                                ]
                        }
                },
**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

MichMich, 2016-10-17:
It has something to do with the "Until" date one of the events has. If you want me to take al look at it, could you send me the calendar feed? michael [at] xonay.com

nigel-daniels, 2016-10-31:
I see the same issue with an iCloud calendar but I don not get a stack trace, the result I get is:

...
Whoops! There was an uncaught exception...
[Error: Invalid UNTIL value: 20160730T115959]
...

I can share my calendar URI if it helps, just let me know a DM route.

MichMich, 2016-11-01:
It's not a stack trace. It's a date which somehow doesn't work for the iCal Library. (20160730T115959)

You can send the calendar URI to michael [at] xonaymedia [dot] nl. Unfortunately I can't give you an estimation on when I can work on this issue, since I currently don't have much time.

nigel-daniels, 2016-11-01:
Ok, np this has been an issue for a while so no hurry.  If you prefer I can open the issue on the iCal library, I just did not realise it was the source of the defect.

berlincount, 2016-12-10:
ical.js seems to be pretty incomple, i.e. RRULE support doesn't implement UNTIL and the like properly. Maybe moving over to https://github.com/tritech/node-icalendar might be sensible?

MichMich, 2016-12-13:
Does node-icalendar also support calendars that are not apple? (google, exchange, etc ... ?)

berlincount, 2016-12-13:
"iCalendar" in this case means the RFC 5545, not the Apple product - so my educated guess is yes.

MichMich, 2016-12-13:
Ok, this might be interesting in the future. Feel free to send a PR.

berlincount, 2016-12-13:
Costed through a lot of the rrule code and forks, and it's a mess. Reasonably simple ical.js using a horribly complex library.

njwilliams, 2017-01-01:
The regular expression in rrule.js requires there to be a trailing 'Z' at the end of the datestring, even though it's not required. If there is no 'Z' at the end of the string then this library will throw the exception. You can fix this by making the Z optional (i.e. add an '?' character after the Z) on line 207 of node_modules/rrule/lib/rrule.js.

MichMich, 2017-01-12:
Continue in #565

closed 2017-01-12 by MichMich