MagicMirrorOrg/MagicMirror #473: Socket.IO missing

Phil795, 2016-10-19:
Hi,

after downloading the master branch, I get some Socket.IO exceptions. 

It look like you forget to include the socket.io folder.

MichMich, 2016-10-19:
Did you run NPM install?

Phil795, 2016-10-19:
Yes of course

MichMich, 2016-10-19:
Can you show me the console output when you run npm start?

Phil795, 2016-10-19:
This is a complete and fresh copy of the project.

I am running npm in version 3.10.3

On Ubuntu 15.10

philipp@UBUNTU:~/Workspace$ git clone https://github.com/MichMich/MagicMirror MagicMirrorCopy
Klone nach 'MagicMirrorCopy' ...
remote: Counting objects: 4302, done.
remote: Compressing objects: 100% (17/17), done.
remote: Total 4302 (delta 4), reused 0 (delta 0), pack-reused 4285
Empfange Objekte: 100% (4302/4302), 10.40 MiB | 3.82 MiB/s, Fertig.
Löse Unterschiede auf: 100% (2252/2252), Fertig.
Prüfe Konnektivität ... Fertig.
philipp@UBUNTU:~/Workspace$ cd MagicMirrorCopy/
philipp@UBUNTU:~/Workspace/MagicMirrorCopy$ npm install
npm WARN deprecated electron-prebuilt@1.4.3: electron-prebuilt has been renamed to electron. For more details, see http://electron.atom.io/blog/2016/08/16/npm-install-electron
npm WARN prefer global coffee-script@1.10.0 should be installed with -g
npm WARN prefer global colorguard@1.2.0 should be installed with -g
npm WARN prefer global jsonlint@1.6.2 should be installed with -g

> electron-prebuilt@1.4.3 postinstall /home/philipp/Workspace/MagicMirrorCopy/node_modules/electron-prebuilt
> node install.js

> magicmirror@2.0.0 prepublish /home/philipp/Workspace/MagicMirrorCopy
> npm run snyk-protect

> magicmirror@2.0.0 snyk-protect /home/philipp/Workspace/MagicMirrorCopy
> snyk protect

Successfully applied Snyk patches
magicmirror@2.0.0 /home/philipp/Workspace/MagicMirrorCopy
├─┬ electron-prebuilt@1.4.3 
│ ├─┬ electron-download@3.0.1 
│ │ ├─┬ fs-extra@0.30.0 
│ │ │ ├── jsonfile@2.4.0 
│ │ │ ├── klaw@1.3.0 
│ │ │ └── rimraf@2.5.4 
│ │ ├── home-path@1.0.3 
│ │ ├── minimist@1.2.0 
│ │ ├─┬ nugget@2.0.1 
│ │ │ ├── pretty-bytes@1.0.4 
│ │ │ ├─┬ progress-stream@1.2.0 
│ │ │ │ ├── speedometer@0.1.4 
│ │ │ │ └─┬ through2@0.2.3 
│ │ │ │   └─┬ xtend@2.1.2 
│ │ │ │     └── object-keys@0.4.0 
│ │ │ ├── single-line-log@1.1.2 
│ │ │ └── throttleit@0.0.2 
│ │ ├── path-exists@2.1.0 
│ │ ├─┬ rc@1.1.6 
│ │ │ ├── deep-extend@0.4.1 
│ │ │ └── ini@1.3.4 
│ │ └── sumchecker@1.2.0 
│ └─┬ extract-zip@1.5.0 
│   ├─┬ concat-stream@1.5.0 
│   │ ├─┬ readable-stream@2.0.6 
│   │ │ └── isarray@1.0.0 
│   │ └── typedarray@0.0.6 
│   ├── debug@0.7.4 
│   ├─┬ mkdirp@0.5.0 
│   │ └── minimist@0.0.8 
│   └─┬ yauzl@2.4.1 
│     └─┬ fd-slicer@1.0.1 
│       └── pend@1.2.0 
├─┬ express@4.14.0 
│ ├─┬ accepts@1.3.3 
│ │ └── negotiator@0.6.1 
│ ├── array-flatten@1.1.1 
│ ├── content-disposition@0.5.1 
[... 606 lines cut ...]
[Wed Oct 19 15:18:43 2016] 127.0.0.1:52800 Invalid request (Unexpected EOF)
[Wed Oct 19 15:18:43 2016] 127.0.0.1:52802 Invalid request (Unexpected EOF)
[Wed Oct 19 15:18:43 2016] 127.0.0.1:52808 [200]: /css/main.css
[Wed Oct 19 15:18:43 2016] 127.0.0.1:52810 [200]: /fonts/roboto.css
[Wed Oct 19 15:18:43 2016] 127.0.0.1:52812 [404]: /socket.io/socket.io.js - No such file or directory
[Wed Oct 19 15:18:43 2016] 127.0.0.1:52814 [200]: /js/defaults.js
[Wed Oct 19 15:18:43 2016] 127.0.0.1:52816 [404]: /config/config.js - No such file or directory
[Wed Oct 19 15:18:43 2016] 127.0.0.1:52818 [200]: /vendor/vendor.js
[Wed Oct 19 15:18:43 2016] 127.0.0.1:52820 [200]: /modules/default/defaultmodules.js
[Wed Oct 19 15:18:43 2016] 127.0.0.1:52822 [200]: /js/logger.js
[Wed Oct 19 15:18:43 2016] 127.0.0.1:52824 [200]: /translations/translations.js
[Wed Oct 19 15:18:43 2016] 127.0.0.1:52826 [200]: /js/translator.js
[Wed Oct 19 15:18:43 2016] 127.0.0.1:52828 [200]: /js/class.js
[Wed Oct 19 15:18:43 2016] 127.0.0.1:52830 [200]: /js/module.js
[Wed Oct 19 15:18:43 2016] 127.0.0.1:52832 [200]: /js/loader.js
[Wed Oct 19 15:18:43 2016] 127.0.0.1:52834 [200]: /js/socketclient.js
[Wed Oct 19 15:18:43 2016] 127.0.0.1:52836 [200]: /js/main.js
[Wed Oct 19 15:18:43 2016] 127.0.0.1:52838 [200]: /fonts/RobotoCondensed-Regular/RobotoCondensed-Regular.woff2
[Wed Oct 19 15:18:43 2016] 127.0.0.1:52840 [404]: /config/config.js - No such file or directory
[Wed Oct 19 15:18:44 2016] 127.0.0.1:52842 [200]: /translations/en.json
[Wed Oct 19 15:18:44 2016] 127.0.0.1:52844 [200]: /translations/en.json
[Wed Oct 19 15:18:44 2016] 127.0.0.1:52846 [200]: /modules/default/helloworld//helloworld.js
[Wed Oct 19 15:18:44 2016] 127.0.0.1:52848 [200]: /css/custom.css
[Wed Oct 19 15:18:44 2016] 127.0.0.1:52850 [200]: /fonts/RobotoCondensed-Regular/RobotoCondensed-Regular.woff2
[Wed Oct 19 15:18:44 2016] 127.0.0.1:52852 [200]: /fonts/Roboto-Thin/Roboto-Thin.woff2

Phil795, 2016-10-19:
I think the problem is grunt 1.*

Socket.IO. requires grunt ~0.4.0

MichMich, 2016-10-19:
Why are you using the php command?

MichMich, 2016-10-19:
Just start the app with: `npm start`. This application does not use php.

Phil795, 2016-10-19:
Ahhh. Thanks man!

Im new in NodeJS, sorry

MichMich, 2016-10-19:
Enjoy. For further support, please use https://forum.magicmirror.builders

closed 2016-10-19 by MichMich