MagicMirrorOrg/MagicMirror #474: Weather config enhancements

zydecat, 2016-10-19:
I would like to suggest a couple of minor enhancements that are configurable.

Firstly, with the Current Weather module, there is the option to display Beaufort Scale wind speeds, or the default data from openweather, which is in metres per second. Would it be possible to add a config option showing wind speed as kilometres per hour? I have forced the change on my mirror on occasion by changing the following variable:

> this.windspeed = this.roundValue((data.wind.speed) * 60 * 60 / 1000) + 'km/h'

Windspeed in km/h makes more sense to me, and wind speed information in Australia is generally given in km/h.

Secondly, with the Weather Forecast module, I'd like to be able to configure the roundValue for the forecast temperatures to zero, rather than leave it at 1. Again, forecast temperatures are generally delivered with no decimal points.

I can go and make these changes in the code of each of the above modules myself, but it is annoying to have to redo the changes if I update the mirror to the latest version.

Anyway, that is my two cents worth.

MichMich, 2016-10-19:
Hi @zydecat, if you make these changes yourself and make them configurable, you can send a Pull Request so I can incorporate it in the next version.

zydecat, 2016-10-19:
Ok I will give that a go. Thanks!

closed 2016-11-08 by MichMich