MagicMirrorOrg/MagicMirror #487: Hey guys! can anyone please help me and tell me how can i run this code on a mobile instead of raspberry pie ?!

DeenABL, 2016-10-26:
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version *and* your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 0.12.13 or later.

**MagicMirror Version**: Now that the versions have split, tell us if you are using the PHP version (v1) or the newer JavaScript version (v2).

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

DeenABL, 2016-10-26:
Hey guys! can anyone please help me and tell me how can i run this code on a mobile instead of raspberry pie ?!

DeenABL, 2016-10-26:
@MichMich can  you please help me and tell me how can i run this code on a mobile instead of raspberry pie ?!

MichMich, 2016-10-27:
For support please visit https://forum.magicmirror.builders. Thanks!

closed 2016-10-27 by MichMich