MagicMirrorOrg/MagicMirror #499: Black screen for copy of hellowold module

titusece, 2016-11-05:
Hello,
I have tried to create a new module in MM, for that I have copied helloworld module from "modules/defauls/helloworld" to "modules/Titus", I just renamed to 'Titus' as a module.

Then I changed the config.js file for a new Titus module.

I am getting black screen now for the Titus module, but just work fine for helloworld.
Why ? Any help ?
Just today I'm started to work on MM.

//Worked
/*
                      {
                      module: 'helloworld',
                      position: 'middle_center',
                      header: 'helloworld module',
                      config: {
                              text: 'I'm helloworld module'
                      } // closes config
                      }, // closes module helloworld
*/
//Not worked
                {
                        module: 'Titus',
                        position: 'middle_center',
                        header: 'Titus module',
                        config: {
                                text: 'I'm Titus module'
                        }
                },

Regards,
Titus S.

MichMich, 2016-11-05:
For support please visit http://forum.magicmirror.builders. Thanks.

closed 2016-11-05 by MichMich

titusece, 2016-11-05:
Thank you, sure.