MagicMirrorOrg/MagicMirror #549: Automatically reloading MM

alexiri, 2016-12-01:
While developing a module, I wanted to be able to automatically reload the server and the browser when I made changes to the code. This was actually pretty easy to achieve, the changes to the code are really simple: https://github.com/alexiri/MagicMirror/compare/master...alexiri:reload

With that done, you can install [supervisor](https://github.com/petruisfan/node-supervisor) and then just run the app like this:

`supervisor -e 'node,js,json,css' serveronly`

This currently doesn't do anything with the electron app, but if there's interest in this feature it could be added.

What do you think? Is this feature interesting for anybody else?

nhubbard, 2016-12-01:
@alexiri I like it. It would be fairly useful in development. Not sure if it would be useful in production, but it would be nice to work with it.

Jopyth, 2016-12-02:
This might be useful for development, but would be bad for production in my opinion. Therefore it should definitely be opt-in.

I am currently writing a modification tool for the `config.js`, (another use case is updating modules), and if the user changes the config.js or updates a module, it might not be expected for them, that the whole app restarts immediately (also this would render the GUI itself temporarily non-functional for a good 10-15 seconds, until the app is restarted). I put this question on the GUI instead, and let the user decide, if they want to restart now, or later.

alexiri, 2016-12-02:
@Jopyth I agree, that's why I added a configuration parameter and set it to false by default.

MichMich, 2016-12-02:
I think the auto restarting should not be part of MM, for this there are perfect solutions like `supervisor`, `pm2` or `nodemon`. What might be interesting though, is to let the frontend automatically reload whenever the backend restarts. This way you can just run a monitored `serveronly` instance, and the frontend will automatically reflect the changed.

This can be easily achieved using the socket messages.

A PR to accomplish this is welcome.

alexiri, 2016-12-02:
I agree, and this is exactly what this commit does. The [reload](https://www.npmjs.com/package/reload) module lets the frontend detect when the backend has restarted and triggers a page reload.

I also suggested ```supervisor``` as the tool to restart MM, but that's independent of this change.

MichMich, 2017-01-12:
pm2 is now part of the install script. pm2 allows auto restarting. (check the docs of pm2 for more info)

closed 2017-01-12 by MichMich