MagicMirrorOrg/MagicMirror #557: MM2 Crashes bluetooth audio

D3rpi, 2016-12-08:
**Platform**: Raspberry PI3 

**Node Version**: 6.9.1

**MagicMirror Version**: V2

**Description**:  Running MM2 in server mode or normal mode crashes bluetooth audio. The bluetooth speaker will randomly stop playing while still being connected. The bluetooth module on the Pi will then become unusable until a reboot. I am using the inbuilt Pi3 bluetooth. 

**Steps to Reproduce**: 1. Connect bluetooth speaker to Pi 3. 
2. Play music to the speaker
3. Launch MM2 in default mode or server mode.
Note: The above order is not important. The issue also arises when connecting the bluetooth speaker when MM2 is already running. 

**Expected Results**: Speaker should keep playing music until stopped by user.

**Actual Results**: Speaker stops playing and bluetooth module becomes unusable. 

**Configuration**: I don't think the issue is related to my config file.

**Additional information**: The crash happens within less than an hour of connecting the bluetooth speaker. Sometimes it happens after being connected for 5 minutes. The longest time without it crashing is 1 hour.

nhubbard, 2016-12-08:
@D3rpi This is leading me to believe that the amount of RAM in use is at fault. Could you try reducing the amount of modules running in your config file?

cc @MichMich

MichMich, 2016-12-08:
The modules don't use too much memory. It's probably Electron that uses a lot of mem.

@D3rpi Could you try running your Pi without MagicMirror for a few hours to check if it's MM related at all? 

How did you configure your memory?

D3rpi, 2016-12-08:
@michmich @nhubbard
I've had this issue for some time now. At first I thought it was something to do with pulseaudio/ bluez timing out the bluetooth speaker. Then I started with a new sd card and had a stable connection without disconnects. I've had the speaker stay connected for more than 24 hours without disconnects, but once I run MM2 the problem comes back. 
Can someone with a bluetooth speaker replicate this issue?

I don't quite follow the question about memory configuration. I also don't think it is a memory issue but will try to monitor the ram and see if there's a spike when the speaker stops playing. Is there a simple way of logging the ram?
I found this which might be helpful. 
http://stackoverflow.com/questions/1868210/how-to-log-the-memory-consumption-on-linux

nhubbard, 2016-12-09:
@D3rpi I actually tried to do something like this. I eventually abandoned it due to the same issues that you were having: Bluetooth was routinely falling asleep.

D3rpi, 2016-12-13:
I logged by ram usage and didn't see a spike when the issue occured.

nhubbard, 2016-12-15:
@D3rpi Yeah, I looked at my notes, and PulseAudio was routinely setting the Bluetooth sink to sleep even if told explicitly not to. I eventually stopped trying after pulling all my hair out and then some (ouch). If you want an alternative, you can use the PiMusicBox distribution to put an interface for audio on top of the MM. It doesn't have Bluetooth, but it *does* have TuneIn Radio. The RPi just isn't the best interface for Bluetooth.

nhubbard, 2016-12-15:
@D3rpi Also, the RPi 3 Bluetooth works, so PulseAudio seems to be the only connected factor.

MichMich, 2017-01-12:
This seems as a PulseAudio to me. Since it's hardware related., there is not much we can do. Feel free to reopen if nessecery.

closed 2017-01-12 by MichMich