MagicMirrorOrg/MagicMirror #561: Cannot display Chinese charactors

yequnsong, 2016-12-11:
Platform: Raspberry Pi 2, Linux raspberrypi 4.4.36-v7+
Nodejs: 6.9.1-1nodes
MagicMirror: JavaScript version (V2)

Everything is ok except the Newsfeed. I configured a Chinese rss feed:
http://news.baidu.com/n?cmd=1&class=technnews&tn=rss
Then the news are displayed with messy code. Looks like now the MM  only support English language. I tried the encoding = 'UTF-8' and 'ISO-8859-1', both of them are displayed with messy code.
Is there any idea on this? Thanks.

stantang986, 2016-12-11:
It's not MM problem -- 百度rss的问题
Change other feeds, chinese will be ok for displaying

nhubbard, 2016-12-11:
@yequnsong I am able to replicate the issue. I *do* get the question mark characters, but even changing the language of the mirror or adding another locale does not change the fact that the characters are not shown properly.

Any thoughts, @MichMich? I'm out of ideas on how to fix this. Could it be our RSS parser?

MichMich, 2016-12-13:
It's probably the font that doesn't have the right characters. Or it's an encoding thing. A bit of research is needed to fix this, preferably by a chinese native. 

A PR is welcome! :)

PengchengWang, 2017-04-06:
It's ok when I add  this url http://news.baidu.com/n?cmd=1&class=technnews&tn=rss with 
`encoding: "gb2312"`
in my config.js ,you can try it again!

closed 2017-09-06 by MichMich