MagicMirrorOrg/MagicMirror #562: Differences between modules default values in the readme and the script file in branch develop

aschulz90, 2016-12-11:
There are some discrepancies between documented default values in the readme and the actual script file for some modules in the develop-branch (maybe master too, haven't checked):

### calendar:

Config Value | Readme | Script-File
------------- | -------- | ----------
Urgency | 0 | 7
titleReplace | None | `{"De verjaardag van ": "",	"'s birthday": ""}`

### alert

Config Value | Readme | Script-File
------------- | -------- | ----------
display_time | 3.5 (says in seconds) | 3500 (comment says in seconds)
welcome_message | Welcome, start was successfull! | false

### currentweather

Config Value | Readme | Script-File
------------- | -------- | ----------
location | "New York" | ""
updateInterval | 300000 (10 minutes) **(actually equals 5 minutes)** | 600000
animationSpeed | 2000 | 1000

### newsfeed

Config Value | Readme | Script-File
------------- | -------- | ----------
feeds | `[ { title: "New York Times", url: "http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml", } ]` |  `{title: "New York Times",	url: "http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml", encoding: "UTF-8" }]`
animationSpeed | 2000 | 2500

### weatherforecast

Config Value | Readme | Script-File
------------- | -------- | ----------
location | "New York" | ""
updateInterval | 300000 (10 minutes) **(actually equals 5 minutes)** | 600000
animationSpeed | 2000 | 1000
initialLoadDelay | 0 | 2500
weatherEndpoint | variable name is `'weatherEndpoint'` | variable name is `'forecastEndpoint'`

I think that should be addressed before 2.1, but I can't say , which one is actually the correct one in every case (most likely just outdated documentation). Therefore i haven't made the changes myself and created no PR, but opened an issue instead.

nhubbard, 2016-12-11:
@aschulz90 If you wish to address this, go ahead! I await your pull request. My recommendation is that you make the README values equal to the script file values. 👍

nhubbard, 2016-12-15:
@MichMich @aschulz90 Should we close this issue now?

PR #563 (2016-12-14): Mm fixes
Fixes #562 

I have updated the READMEs with the values defined in the script-files, EXCEPT for the calendar module, where `titleReplace` was defined as this:

```
titleReplace: {
    "De verjaardag van ": "",
    "'s birthday": ""
}
```

I don't think that makes sense, that this is the default value for it (in respect to the default defined calendar) and changed it to an empty object (`{}`) instead.
closed 2016-12-19 by MichMich