MagicMirrorOrg/MagicMirror #567: NPM warnings when running npm install.

MichMich, 2016-12-13:
When installing the node dependencies, the installer ends with a warning:

````
npm WARN grunt-yamllint@0.2.0 requires a peer of grunt@~0.4.0 but none was installed.
npm WARN stylelint-config-standard@15.0.0 requires a peer of stylelint@^7.6.0 but none was installed.
````

This resulted in a lot of people thinking they were unable to install the application. 

@nhubbard  Since this is a direct result of the linters, is this something you can look into? A PR solving this issue would be very welcome. (And I'd like to include this fix in the release of 2.1.0)

Thanks!

nhubbard, 2016-12-13:
@MichMich This is a slight concern of mine. From what it appears, the packages have peer dependencies, which npm versions greater than 2 will warn you about. To fix this issue, all our npm install commands need to have the `--loglevel error` flag on them, which according to [here](https://docs.npmjs.com/misc/config) implies that only errors will trigger logging. Either that, or we can add an install script that just prints "Install successful!" in green to let the user know that the system is all good. Either one is your choice! I welcome the use of either or both in conjunction. :+1:

MichMich, 2016-12-14:
Send me a PR with your personal favourite ;) The loglevel seems like a good idea. For most users, the warnings are unimportant.

MichMich, 2017-01-10:
@nhubbard Still willing to fix this? :)

nhubbard, 2017-01-11:
@MichMich Sorry, exams have got me looking left, right and center on studying. I'll try to get to it this weekend if I have time.

closed 2017-01-15 by MichMich

roramirez, 2017-03-05:
Note:  This warning should disappear with the last update of grunt-yamllint
https://github.com/geedew/grunt-yamllint/commit/55652f65e7d112ff23d34eb6e6dc4d6166b719b4