MagicMirrorOrg/MagicMirror #568: wrong age in calendar module...

chamaeleon82, 2016-12-13:
I´ve created google calendar as a birthday calendar. Birthday is created as an appointment (whole day, repeat every year). Start date is the birthday and year of birth. 
My problem is that the display of age is a year less. So instead of 35 years old only 34 is displayed.

Does anyone know this problem?

MichMich, 2016-12-14:
That's odd. I suggest discussing this issue on the forum: http://forum.magicmirror.builders

henglert, 2016-12-28:
As I wrote in the forum: 

`--- a/modules/default/calendar/calendar.js
+++ b/modules/default/calendar/calendar.js
@@ -126,7 +126,7 @@ Module.register("calendar",{
                                repeatingCountTitle = this.countTitleForUrl(event.url);

                                if(repeatingCountTitle !== '') {
-                                       var thisYear = new Date().getFullYear(),
+                                       var thisYear = new Date(parseInt(event.startDate)).getFullYear(),
                                                yearDiff = thisYear - event.firstYear;

                                        repeatingCountTitle = ', '+ yearDiff + '. ' + repeatingCountTitle;
`

MichMich, 2016-12-29:
Please send  PR. Thanks!

roramirez, 2017-01-06:
Hey @MichMich  this issue need be closed.

closed 2017-01-06 by chamaeleon82